vitormhenrique/OctoPrint-Enclosure #1: Other Temperature Sensors

wiesel24, 2016-09-30:
Is there any possibility to change the sensor type to DS18B20? So DHT sensors are sometimes hard to get in some countries. Would be nice if there is an easy way to change this.

Thanks in advance
Regards
Dominik

vitormhenrique, 2016-10-10:
It should be implemented now.

wiesel24, 2016-10-13:
Hi, 
Wow, thank you for your afford! I tried it but it didn't work. I edited the GetTemperature.py script to take a look and the correct ID of my DS18B20 sensor is in it. Sorry for not understanding the code, but it  have a .py script that works and the temperature is displayed. So the temperature sensor is correctly installed. I don't get any output if I ran your script. I think that's the problem. 

And sorry for begging, but it is also possible to address a second Fan in the Plugin?

Maybe you will find a solution. 

Thank you!

Regards 
Dominik

vitormhenrique, 2016-10-13:
You should use GetTemperature1820.py instead of GetTemperature.py.
Manually copy the script on "~/.octoprint/plugins/OctoPrint-Enclosure/extras/"  or any other place.

Make it executable
` chmod +x ~/.octoprint/plugins/OctoPrint-Enclosure/extras/GetTemperature1820.py
`
 Also the script is waiting for a argument, type
 `~/.octoprint/plugins/OctoPrint-Enclosure/extras/GetTemperature1820.py 1820`

And see if the terminal print the temperature. Also please get the my updates from yesterday.

DomLuu, 2016-10-13:
Hallo Dominik 

gehe davon aus du sprichst auch Deutsch (wegen deinem Nick). Ich habe Vito gebeten den DS1820 zu implementieren, da ich mein eigenes Script dazu nicht eingebunden bekommen habe. Vitor hat entsprechend gleich alles in sein Plugin implementiert. Ich würde dir folgendes Vorgehen vorschlagen:
1) Enclosure Plugin über Octopi zu deinstallieren.
2) den Ordnerpfad per Console manuell zu erstellen: (mkdir ...) /.octoprint/plugins/OctoPrint-Enclosure/extras
3) alle 3 Get..py Files manuell reinzukopieren (https://github.com/vitormhenrique/OctoPrint-Enclosure/tree/master/extras)
4) alle Files ausführbar machen: chmod +x ~/.octoprint/plugins/OctoPrint-Enclosure/extras/GetTemperature1820.py 
5) wiringPi (falls noch nicht vorhanden zu Installieren) / Achtung bei mir hat es einen zusätzlichen Eintrag in der /etc/rc.local benötigt  (da muss ganz unten noch "dtoverlay=w1-gpio" nein
6) Plugin NEU installieren
7) Die Enclosure PluginSettings entsprechend anzupassen

Grüsse Dominik

wiesel24, 2016-10-16:
Hi Vitor, Hallo Dominik :-)

@Vitor 
Thank you everything worked perfect after your and dominik's advices!!! I don't want to be nervy but maybe if you got time some day maybe there is the possibility to add a second fan option to this superb plugin. I know you're doing this in your free time! Thank you for everything Vitor! 

@Dominik
Danke deiner tollen Anleitung habe ich es endlich hinbekommen. Lustig das du auch zufällig Dominik heißt  ;-)
Vielleicht kann Vitor ja wirklich irgendwann einen 2 Fan dazu basteln im Interface. 
Danke nochmals!

BG
Dominik

vitormhenrique, 2016-10-18:
I'm current working on an alternative to OctopiPanel. After I finish that I can implement the second fan. You should be able to implement that fairly easily using my current code.

closed 2016-10-18 by vitormhenrique

vitormhenrique, 2016-12-10:
Dominik,
Just to let you know that I updated the code to enable control of 4 generic IO on rPi. You can now control up to 4 different things using the plugin.

DomLuu, 2016-12-10:
Hi vitor 

I just finished my Enclosure last night :-)
I got without a heater 45C and so I have to control a fan for cooling. 

Nice work with the new version.... I will try it!

My Enclosure:

Sent by iPhone5S

DomLuu, 2016-12-10:
Hi Vitor

the new version looks great…. thx

Just one thing, the old version disable the light after the print job is finish.
Is there any possibility to reactivate this „disable Light“ after Printjob?

Thx my friend…

vitormhenrique, 2016-12-10:
Yeap...

I'll add a "disable this IO after job is complete " on the settings....  i'll also add a possible delay...  

I'll use the plugin to shutdown the entire printer after printing. 

Atenciosamente,
Vitor Henrique

vitormhenrique, 2017-01-01:
I added the functionality to disable the IO when print is complete, you can also configure a delay if you want. 
There is also a new functionality, a filament sensor, it will pause your print if you run out of filament.

DomLuu, 2017-01-01:
wow thats great - thx soon much and a happy new year!