vitormhenrique/OctoPrint-Enclosure #2: OctoPi Install

Tom2510, 2016-11-01:
I have attached and read temperature from my 2302 fine from the command line but the Octoprint interface shows no data.

I'm guessing this is because when I try and add the executable flag to the python scripts I get a file not found error, checking manually over SSH the plugins folder is empty... The plugin shows up in the web interface fine.

Im guessing this is because i used the prebuilt Octopi image? Can you help please!

vitormhenrique, 2016-11-01:
You need to manually copy the scripts on your octopi image! 

You can try using the following commands, after you ssh to your rpi:

`
mkdir -p ~/.octoprint/plugins/OctoPrint-Enclosure/extras/
cd ~/.octoprint/plugins/OctoPrint-Enclosure/extras/
wget https://raw.githubusercontent.com/vitormhenrique/OctoPrint-Enclosure/master/extras/GetHumidity.py
wget https://raw.githubusercontent.com/vitormhenrique/OctoPrint-Enclosure/master/extras/GetTemperature.py
chmod +x ~/.octoprint/plugins/OctoPrint-Enclosure/extras/GetTemperature.py
chmod +x ~/.octoprint/plugins/OctoPrint-Enclosure/extras/GetHumidity.py
`

After this try testing the scripts again...

`./GetTemperature.py 2302 4`

Tom2510, 2016-11-01:
Fantastic, worked perfectly, thanks for you help and the perfect plugin for my enclosure 👍

vitormhenrique, 2016-11-01:
No problem, I updated the readme file to clarify the installation process :)

closed 2016-11-01 by vitormhenrique