vitormhenrique/OctoPrint-Enclosure #3: Heater disabled but getting log errors

flyer007, 2017-02-03:
2017-02-03 15:07:06,556 - octoprint.plugins.enclosure - INFO - Failed to convert to float
2017-02-03 15:07:06,612 - octoprint.plugins.enclosure - INFO - Failed to convert to float
2017-02-03 15:07:06,615 - octoprint.plugins.enclosure - INFO - Turning heater off.
2017-02-03 15:07:16,672 - octoprint.plugins.enclosure - INFO - Failed to convert to float
2017-02-03 15:07:16,725 - octoprint.plugins.enclosure - INFO - Failed to convert to float
2017-02-03 15:07:16,727 - octoprint.plugins.enclosure - INFO - Turning heater off.
2017-02-03 15:07:26,783 - octoprint.plugins.enclosure - INFO - Failed to convert to float
2017-02-03 15:07:26,839 - octoprint.plugins.enclosure - INFO - Failed to convert to float
2017-02-03 15:07:26,843 - octoprint.plugins.enclosure - INFO - Turning heater off.
2017-02-03 15:07:36,899 - octoprint.plugins.enclosure - INFO - Failed to convert to float
2017-02-03 15:07:36,957 - octoprint.plugins.enclosure - INFO - Failed to convert to float
2017-02-03 15:07:36,961 - octoprint.plugins.enclosure - INFO - Turning heater off.
2017-02-03 15:07:46,993 - octoprint.plugins.enclosure - INFO - Failed to convert to float
2017-02-03 15:07:47,024 - octoprint.plugins.enclosure - INFO - Failed to convert to float
2017-02-03 15:07:47,029 - octoprint.plugins.enclosure - INFO - Turning heater off.
2017-02-03 15:07:57,085 - octoprint.plugins.enclosure - INFO - Failed to convert to float
2017-02-03 15:07:57,141 - octoprint.plugins.enclosure - INFO - Failed to convert to float
2017-02-03 15:07:57,145 - octoprint.plugins.enclosure - INFO - Turning heater off.
2017-02-03 15:08:07,201 - octoprint.plugins.enclosure - INFO - Failed to convert to float
2017-02-03 15:08:07,256 - octoprint.plugins.enclosure - INFO - Failed to convert to float
2017-02-03 15:08:07,260 - octoprint.plugins.enclosure - INFO - Turning heater off.
2017-02-03 15:08:17,316 - octoprint.plugins.enclosure - INFO - Failed to convert to float
2017-02-03 15:08:17,372 - octoprint.plugins.enclosure - INFO - Failed to convert to float
2017-02-03 15:08:17,376 - octoprint.plugins.enclosure - INFO - Turning heater off.

vitormhenrique, 2017-02-03:
yeap... even with the heater disabled it is going to try to read the temperature. Should I create a separate check box for enabling disabling the temperature sensor?

flyer007, 2017-02-03:
It would save log space. 
BTW.....great work on this!  Have you thought about adding PWM to the GPIO?  For RGB control or dimming?

vitormhenrique, 2017-02-03:
I will only log this if the heater is enable, that will accomplish debugging only when needed. As I'm using RPi.GPIO, so I guess is pretty simple enabling PWM, I'll take a look o the following revision. I have new features requested. 
Next version will include: 
- email notifications (if filament sensor is enable and if run out of filament) 
- turning on automatically gpio when printer started (configurable)
- adding buttons to control IO on the control tab instead of the enclosure tab.
- I'll change the logging feature to save log space

vitormhenrique, 2017-02-03:
I'll leave this issue open until I publish the new code.

vitormhenrique, 2017-03-01:
Closing this issue, I improved the debug information on 1.6 release.

closed 2017-03-01 by vitormhenrique