vitormhenrique/OctoPrint-Enclosure #4: Failed to convert to float using DS18B20?

pcbroch, 2017-02-09:
I'm getting an error in my logs that says "octoprint.plugins.enclosure - INFO - Failed to convert to float" when trying to read the temperature from a DS18B20 sensor.  If I run the scrip manually, I get the proper temperature:

pi@octopi:~/.octoprint/plugins/OctoPrint-Enclosure/extras $ sudo ~/.octoprint/plugins/OctoPrint-Enclosure/extras/GetTemperature1820.py 1820
20.3

It seems sTemp is empty after the Popen to the get temperature script... 

Anything I'm missing?

vitormhenrique, 2017-02-09:
is that the only error that is being logged? Also, does the temperature shows up on the enclosure tab?

pcbroch, 2017-02-09:
Yes, that's the only error.  The temperature shows as 0 in the enclosure tab.

vitormhenrique, 2017-02-09:
Can you post the entire log, and please confirm that you have set the correct temperature sensor under the plugin settings.

pcbroch, 2017-02-10:
I am a moron.  I was calling GetTemperature.py and not GetTemperature1820.py

So sorry for wasting your time...

closed 2017-02-10 by pcbroch

vitormhenrique, 2017-02-10:
no worries! glad that you got it working!