vitormhenrique/OctoPrint-Enclosure #6: After upgrading, I no longer see the temp on my DS18B20

jdebriyn, 2017-03-12:
I checked the sensor by running through the directions at: (https://learn.adafruit.com/adafruits-raspberry-pi-lesson-11-ds18b20-temperature-sensing?view=all). The sensor reads fine using those directions.

vitormhenrique, 2017-03-12:
Could you please remove and install the plugin again using the github link:
https://github.com/vitormhenrique/OctoPrint-Enclosure/archive/master.zip
After that do the following checks:
check the setting screen and see if everything is correct, it should have 1820 on the sensor type and the location of the script should point to the correct script "GetTemperature1820.py"
If that is ok, ssh on your raspberry pi and type the following commands
`cd ~/.octoprint/plugins/OctoPrint-Enclosure/extras/
GetTemperature1820.py 1820`

Temperature should show on the terminal. Let me know the results so I can help you trouble shoot the issue.

jdebriyn, 2017-03-13:
Thank you for the quick reply. I just uninstalled and reinstalled.

When I execute:
`GetTemperature1820.py 1820`
I get
> -bash: GetTemperature1820.py: command not found

If I check the directory with:
`dir`

I get:

> GetTemperature1820.py

Everything looks like it it there.

vitormhenrique, 2017-03-13:
Try to make the script executable chmod +x GetTemperature1820.py
And I'm sorry you need to execute the script like this:

`./GetTemperature1820.py 1820`

jdebriyn, 2017-03-13:
That worked. I got "26.4." Still nothing on the UI though. :-(
[image]

[image]

vitormhenrique, 2017-03-13:
That is ok, just step by step trouble shooting... 
now ssh into your raspberry pi and stop the service using:
`sudo service octoprint stop`
After that, start the octoprint manually using:
`/home/pi/oprint/bin/octoprint`

Go to the plugin UI setting screen, on advanced and enable debug mode. Take a look on what my enclosure is logging and let me know....

jdebriyn, 2017-03-13:
Not a ton in the log. I turned the heater function on and off to get something to show up.

> 2017-03-13 02:17:19,607 - octoprint.plugins.enclosure - INFO - DEBUG -> Seting enclosure temperature: 0
2017-03-13 02:17:19,612 - octoprint.plugins.enclosure - INFO - DEBUG -> Current heater status: False previous heater status False

jdebriyn, 2017-03-13:
More log entries after clicking around:

> 2017-03-13 02:22:45,405 - octoprint.plugins.enclosure - INFO - DEBUG -> Seting enclosure temperature: 50
2017-03-13 02:22:45,410 - octoprint.plugins.enclosure - INFO - DEBUG -> Current heater status: True previous heater status False
2017-03-13 02:22:45,412 - octoprint.plugins.enclosure - INFO - Turning heater on.

jdebriyn, 2017-03-13:
Sorry, I also followed the manual restart steps you provided. Thank you again for debugging with me.

vitormhenrique, 2017-03-13:
Yeap... I just figure it out that I had introduced a bug on that release, sorry about the trouble! Update the plugin and it should work fine now, just tested with a brand new raspberry pi, plugged the 1820 sensor and it is working now.

vitormhenrique, 2017-03-13:
It was giving an error before like this:
`UnboundLocalError: local variable 'sHum' referenced before assignment`

jdebriyn, 2017-03-13:
I updated about 2 hours ago (it popped up during our conversation). I'm on 2.3 according to the plugin manager. Is there a newer version?

vitormhenrique, 2017-03-13:
yeap... 2.4 is just out of the oven!

jdebriyn, 2017-03-13:
Perfect. One more noob question: Can I update manually? or do I have to wait for octoprint to to notify me?

Also, may I request a feature? I love the heater option, but I would also like a fan option that is temperature controlled. Basically, the same as the heater, but kicks on at the set temp. (Maybe I can use the heater in reverse?)

vitormhenrique, 2017-03-13:
yeap, you can go to the plugin manager, and check for updates.

Did it work for you? Multiple people requested the cooling feature, I guess people who would use the cooling it would not use the heater right?

sajovicd, 2017-03-13:
Or in the winter will use heater,in the summer will use cooler 😉.

vitormhenrique, 2017-03-13:
no problem, I just don't see why using both at the same time as they would fight to get the temperature on opposite directions. I'll implement the cooling option. :)

Just let me know if the bug was fixed for you so I can close this issue.

jdebriyn, 2017-03-13:
It still says everything is up to date. I will check again tonight. If that doesn't work, I will uninstall/reinstall.

[image]

sajovicd, 2017-03-13:
Try with advance button and force upgrade check.
I had same situation.

jdebriyn, 2017-03-13:
@sajovicd Thank you. I didn't even see the "Advanced" button. Everything is working now. Feel free to close out this ticket and thank you for all the help and the great plugin!

vitormhenrique, 2017-03-14:
Cool, thanks for testing,

closed 2017-03-14 by vitormhenrique