vitormhenrique/OctoPrint-Enclosure #8: Add support for dual extrusion filament sensing.

zorrobyte, 2017-03-14:


vitormhenrique, 2017-03-14:
No problem, this would be fairly easy to implement, I'll add together with the cooling option that some users requested on the 2.5 version. 
If I have enough free time I'll try to code that this weekend.

Sodamaexchen, 2017-03-17:
Can you make it with 4 inputs ?
I thinking about the Multimaterial Upgrade from Prusa.

vitormhenrique, 2017-03-17:
I was thinking a little bit more on this, It is definitely easy to implement to use raspberry pi, but maybe a better solution, and generic for as many sensors that you want, is to use a logic [gate](https://en.wikipedia.org/wiki/NAND_logic#AND) and "mix" all outputs to one output that is processed by raspberry pi. Implementing this very is very simple, for active low sensors, just using a [SN74HC00N](http://www.digikey.com/product-detail/en/texas-instruments/SN74HC00N/296-1563-5-ND/277209) would do the trick... The advantage of this is to use only one IO from raspberry pi. What do you guys think?

zorrobyte, 2017-03-17:
I have no problem doing that @vitormhenrique . It would surely decrease code complexity. Thank you for the component link!

What do you think @Sodamaexchen ?

vitormhenrique, 2017-03-17:
It is not even about the code, adding more filament inputs it's basically copy and paste. I just think it is more elegant.

Sodamaexchen, 2017-03-17:
Hey guys,

some times the solution is quite simple. 
We even do not need a logic gate, just wire the Sensors in parallel or serial, depending if you have NC or NO.
So for me the hardware solution is OK.
But i have some more advanced enhancement requests, but i will start a new issue for them.

BTW. this plugin i was waiting for.

vitormhenrique, 2017-04-11:
Implemented on version 3.0 that is on dev branch right now, if you want to test it out uninstall the current plugin and install it again manually using the development branch. I still need to do more testing, all architecture of the software changed, so it might have few bugs.
 https://github.com/vitormhenrique/OctoPrint-Enclosure/archive/dev.zip

closed 2017-04-14 by vitormhenrique