vitormhenrique/OctoPrint-Enclosure #9: Setting Enclosure Temperatur via G-Code / Octoprint

Sodamaexchen, 2017-03-17:
I have this crazy idea to set the enclosure temperatur depending on the filament type im using.
And not simple switch the fan on/off but using PWM with a PID control loop. (Dont have a heater, too scary)

I build a fan controller with an Arduino where i can set the temp and the fan speeds up if the temp gets too high. But it is missing the link to Octoprint.

I am ok with mixing examples for the arduino and make them run with a good looking code, but Linux on the Pi is something i have trouble with.

As a first step setting the temp in Octoprint would be great.

vitormhenrique, 2017-03-17:
I'll add the cooling option on my next update, it is a feature that is being requested for a while now, there is probably no need to PID control it, because the residence time (time needed to change the variable, on this case, the enclosure temperature) is so high, at least a bunch of seconds. So basically a proportional control, with the fan turning on 100% is more than enough to have a stable temperature inside the enclosure, this is basically what I'm doing for the heater and works perfectly. 
To properly use PID control the fan, you would need to characterize your enclosure thermal properties and some how calculate parameters, I don't think this would have any gain against the simple solution.
It would, however, be a good option to allow the IO to be controlled by PWM in general, so whenever you turn on, it would use the settings for that IO, and drive it on that frequency. I can add this on the next release as well. 
I'll probably change the architecture of the plugin a little bit. The implementation of the cooling feature will be as such as you either choose to cool your enclosure or you heat it, you will not be able to use both at the same time.

vitormhenrique, 2017-03-17:
I missed the G-Code title, I can intercept g-code commands on the plugin, I guess I can intercept M141 gcode and turn heater with that.

Sodamaexchen, 2017-03-17:
if i can set it from Octoprint its ok, because i don't have no idea how to get the slicer to produce the neccesarry gcode

DevKitchen, 2017-04-18:
I'd quite like PWM fan control too if possible, would be useful to be able to adjust my fans even though they aren't used for cooling, more for heat redistribution really.

mcmasterp, 2017-04-18:
I only use it to control relays. I hadn't thought of pwm. Can the pi pins run a fan directly?

vitormhenrique, 2017-04-18:
Relays don't like PWM, you would need to use some type of transistor (MOSFET) to control the output.
But how do you guys envision this?
Would the PWM (percent value) be controlled on the enclosure plugin tab? or on the setting?

mcmasterp, 2017-04-18:
Settings is best but some sort of adjusting from tab would be slick. Maybe a choice in setting to allow editing in tab? I'm yet to envision a use but I'm sure I'll come up with one. The idea of moving air around is interesting

vitormhenrique, 2018-03-01:
I just finished re-writing most of the plugin and **implemented this requested** feature, tested 95% of the plugin.

If you want to test it out uninstall the current plugin, go to the plugin manager, select get more, and where it says "from URL" use the development branch:

https://github.com/vitormhenrique/OctoPrint-Enclosure/archive/dev.zip

More documentation here:
https://github.com/vitormhenrique/OctoPrint-Enclosure/tree/dev

I'll push this to production after more people test it out.

vitormhenrique, 2018-03-05:
g-code control documented here: https://github.com/vitormhenrique/OctoPrint-Enclosure/wiki/G-CODE-Control

vitormhenrique, 2018-03-07:
Implemented on V4.0

closed 2018-03-07 by vitormhenrique

LDXChrisS, 2019-03-21:
I've just started to use your plugin to help control my enclosure temperature so I can heat it for ABS and cool it for PLA. I have a couple of questions...

1) Did you implement this via M141 or only thru the ENC gcode? (looking thru the source just looks like ENC)

2) How best to configure this for using a heater for ABS and a cooling fan for PLA ?

vitormhenrique, 2019-03-21:
1. It only support ENC gcodes or setting it using the UI. 

2. Every enclosure / printer is different, for very large prints i have a personal heater inside of the enclosure and I set the enclosure to 45C.  For PLA I had a fan constantly blowing inside of the enclosure not really controlling temperature, but I found that that was really not needed so I don't use anything anymore. So it will vary on your setup.

spiff72, 2020-02-01:
I know this is an older closed issue/feature request, but I have a question related to this.

I have a temperature sensor in my enclosure and a PWM controlled fan output and that is all working well with the "link temp to PWM" option.  I set the lower limit of my target temp to say 27C and upper limit to 30C, and the associated duty cycles are 50 and 100 for these values.

The trick is I only want to do this for PLA prints.  For ABS or ASA or PETG, I would like to use higher set temps.  Is there a way to do this "automagically" via reading the GCODE?  Maybe if it sees that the extruder temp is set for between 200 and 225C, it assumes PLA, and if extruder temp is set higher, it can assume PETG/ABS/ASA?  (Or maybe the same could be done with the bed temperature?  (60C = PLA, 80C or higher = PETG/ABS/ASA)?

Or if I did this with the ENC gcodes mentioned above, is it possible to put if/then statements in GCODE that would set this correctly?  I haven't tinkered with GCODE much, and that is probably my next research step.   If I took this approach, what would be the proper ENC commands to set the Temperature A, Duty A, Temperature B, Duty B values?

If it matters, I am using a Prusa i3 MK3S with the latest PrusaSlicer.

Thanks!