vitormhenrique/OctoPrint-Enclosure #10: How about some safety code, a flame or smoke detector that sounds an alarm and or shuts down the main power supply.

DilligafBUZZ, 2017-03-19:


vitormhenrique, 2017-03-20:
I think, by the time that any sensor detects fire, it is to late to turn the power down, the fire will continue with or without it. We could ad an alarm but my opinion is that you need a proper and tested fire alarm installed on your house. I don't want to be responsible for any house burning because the plugin did not detect a fire.

DilligafBUZZ, 2017-03-20:
we have  working tested and fully functioning smoke detectors in the house, but a detection system inside the cabinet would have allerted me sooner than the house alarms, and saved us a lot of damage. 
I will look into installing a house standard smoke detector inside the cabinet. thanks for your input

closed 2017-03-20 by DilligafBUZZ

vitormhenrique, 2017-03-20:
I can definitely write something, I just don't want to be liable, I can make it generic, like a event based system something like: if this pin is high, do some action, I could add the filament detection inside of it, but this will be on a lower priority. I'm re-writing the plugin to make it more generic and capable.

DilligafBUZZ, 2017-03-20:
Thank you that would be very helpful, 
I do appreciate that a properly installed house smoke detector installed in your house is a no-brainer and a smoke/flame detector inside the cabinet is in no way a substitute for a properly installed smoke detector in your home. but something running in the background in the printer that shut down the mains power or a least shutting down the cabinet fan (not giving the fire more oxygen) would help in reducing the damage caused by a fire inside the cabinet.
I look forward to your improvements to your already great addon.

mcmasterp, 2017-03-27:
Yes I am also worried and I have landed on the idea of 16g co2 cartridge attached to a solenoid valve, attached to a relay board, attached to the pi. it was shockingly cheap on amazon and looks so cool lol. but with the events system this would LIKELY extinguish a small fire. combined with a relay to remove power, you have a way safer system

vitormhenrique, 2017-04-11:
Implemented some sort of event trigger that can be used for safety, please check the dev branch.