vitormhenrique/OctoPrint-Enclosure #12: trigger action on temp

mcmasterp, 2017-03-23:
The fan and other io being able to react to temperature readings, gpio triggers, web post\get etc.

vitormhenrique, 2017-03-23:
Cooling will be added on the next release, I guess that fan reacting to the temperature is what you expect.

Event based triggers is on my TODO list, but are supper generic and I'll only implement on the future, I still haven't thought on a way that is easy to create the events and generic enough to allow good customization, if you have idea on how the setting UI should behave give me your insights.

I have no idea what do you mean by web post \ get, and what you want to do with it. 

Please if you have a feature request, be more clear on what you expect, and how it should behave.

mcmasterp, 2017-03-23:
Hello, thank you for the response and work in general. I was typing on my phone at the time so I kept it short.

My enhancement idea is to add the ability to control the io's based on actions like a gpio pin going high, or the temp sensor reaching a high or low, or in response to web requests via simple POST or GET url structures.

I have an oversized reprap i2 samuel that I built a 2ft square wooden enclosure for. I would like to be able to control certain features automatically in addition to just add flexibility for future ideas. 

I am attempting to control:
1) an enclosure fan for cooling
2) an led light strip
3) the power supply
4) a solenoid for fire suppression
5) Ramps fan
6) hotend fan

Obviously cooling is coming you said, psu control is covered by another plugin, and lights/ramps fan/hotend fan can be accomplished already. That leaves my idea for fire suppression which is a 16g co2 cartridge connected to a 12v solenoid, I'd like to have it release the co2 when the temp gets critical, that's why I asked for the temp triggering of a gpio

Let me know if that all makes sense
an enclosure fan based on temps sensed by the temp+hum sensor.

vitormhenrique, 2017-04-11:
Implemented on version 3.0 that is on dev branch right now, if you want to test it out uninstall the current plugin and install it again manually using the development branch. I still need to do more testing, all architecture of the software changed, so it might have few bugs.
 https://github.com/vitormhenrique/OctoPrint-Enclosure/archive/dev.zip

closed 2017-04-14 by vitormhenrique