vitormhenrique/OctoPrint-Enclosure #13: REQUEST: Only call GPIO.cleanup() for locally configured pins.

kantlivelong, 2017-03-24:
This is in reference to [OctoPrint-PSUControl#19](https://github.com/kantlivelong/OctoPrint-PSUControl/issues/19)

It appears that this plugin uses GPIO.cleanup() with no arguments which can break GPIO for other OctoPrint plugins. Instead configured GPIO pins should be tracked locally and individually cleaned up when necessary.

mcmasterp, 2017-03-24:
Haha i just came here to report this as well.

kantlivelong, 2017-03-24:
Figured I'd save you the effort of explaining it. :)

vitormhenrique, 2017-03-24:
Yeap, I'm aware of this... the problem is that to change between board and bcm pin numbers I still need to do a global GPIO.cleanup(), and this plugin has the option to change between the two... how do you propose to fix this?

kantlivelong, 2017-03-24:
I recently added the option to specify BOARD vs BCM to PSUControl. Basically it will try to configure for the specified mode but if already set it will use the existing mode while converting the pin numbering.

See https://github.com/kantlivelong/OctoPrint-PSUControl/blob/devel/octoprint_psucontrol/__init__.py#L150

vitormhenrique, 2017-03-24:
I though about doing that, but it is just so "not polished"... imagine in 5 revisions, you would need to maintain the mapping for every single one... there must be a better way... 
granted that it is better than having the plugin breaking other plugins....

vitormhenrique, 2017-03-24:
I actually going to do the following: if there is already a specification on the mode type, I'll prevent the user from changing it, so it is the user responsibility to get the correct value.

kantlivelong, 2017-03-24:
I thought about that as well and found that RPi.GPIO actually has the table hardcoded the same way. So if there is another revision of the pinout then RPi.GPIO would be broken anyway.

vitormhenrique, 2017-04-11:
Implemented on version 3.0 that is on dev branch right now, if you want to test it out uninstall the current plugin and install it again manually using the development branch. I still need to do more testing, all architecture of the software changed, so it might have few bugs.
 https://github.com/vitormhenrique/OctoPrint-Enclosure/archive/dev.zip

vitormhenrique, 2017-04-14:
removed the GPIO.cleanup() from the code, added check to see if the GPIO mode was set previously, and forced the GPIO on those cases. 
Also added a button to force clean the board mode for advanced users that might want to change the mode

closed 2017-04-14 by vitormhenrique