vitormhenrique/OctoPrint-Enclosure #14: failed to read temp and hum

mcmasterp, 2017-03-24:
Me again. I'm getting failed to read errors, I use a dht11 on pin 4 (or 7 however you look at it).

I enabled debug and attaching relevent bits:

`2017-03-24 01:57:36,623 - octoprint.plugins.enclosure - INFO - DEBUG -> Reading temperature stdout: <open file '<fdopen>', mode 'rb' at 0x6a0aae38>
2017-03-24 01:57:36,625 - octoprint.plugins.enclosure - INFO - Failed to read Temperature
2017-03-24 01:57:36,681 - octoprint.plugins.enclosure - INFO - DEBUG -> Reading humidity stdout: <open file '<fdopen>', mode 'rb' at 0x6a0aaee8>
2017-03-24 01:57:36,683 - octoprint.plugins.enclosure - INFO - Failed to read Humidity`

Let me know anything else you need to identify the conflict.

vitormhenrique, 2017-03-24:
Is the temperature being displayed? I mention on the read-me that dht sensors can't work reliably on linux, due to the kernel can give proper priority to gpio.
They will faill a bunch of times, but on the overall scheme they should still be usable.

mcmasterp, 2017-03-24:
No it only shows zeros

vitormhenrique, 2017-03-24:
Follow the instructions on adafruit website that i liked and check if your wiring / sensor is working. If you can get them working outside the enclosure I can help you trouble shooting...

mcmasterp, 2017-03-25:
I will as soon as I get a new pi3. my 3.3v regulator I use to sense the 12v psu decided to crap out and dump 12v into the pi

mcmasterp, 2017-03-25:
oh man I missed that whole setup instructions for the dht11. I did zero of those steps of installing software and libraries. When I get the new pi I'll know for sure but I'm hopeful it's just my stupidity. You may need to link to the instructions in the description in settings to avoid the iD10t error

vitormhenrique, 2017-03-25:
I'll do that, I'll close this issue for now, sorry to hear about your rpi...  Try using opto isolators next time with the 12V, that will avoid those type of issues.

mcmasterp, 2017-03-25:
new pi arrives tomorrow.

I know,  i even have a pile of optos i was too lazy to add to the 1117 regulator, i'll report back tomorrow

mcmasterp, 2017-03-27:
Ok new pi up and running. I have for now uninstalled PSU control plugin until a later date. I installed the dht11 stuff, the instructions were perfect. reboot and I at first saw blank spaces which gave me hope as before only zeros but sadly it within a minute went to zeros and no change since. 

Is there a way to use a thermosister? I got lots of those lol. maybe on ramps boards and others there is an unused temp sensor pins that could be used to report the enclosure temp?

mcmasterp, 2017-03-27:
I tried the examples from adafruit and get back:

`pi@octopi:~/Adafruit_Python_DHT/examples $ sudo ./AdafruitDHT.py 11 4
Temp=10.0*  Humidity=150.0%
pi@octopi:~/Adafruit_Python_DHT/examples $ sudo ./AdafruitDHT.py 11 4
Temp=10.0*  Humidity=148.0%
`

So I THINK that means it's working on the pi?

vitormhenrique, 2017-03-27:
Did you copy the scripts from the github page on your raspberry pi? 
If so, log in your rpi and do the following:

`
cd  ~/.octoprint/plugins/OctoPrint-Enclosure/extras/
./GetTemperature.py 11 4
`

mcmasterp, 2017-03-27:
scripts were copied yes. I ran the script you asked and it returned 9.0
the same idea but with gethumidity.py returns 145.00

Now that I ran them I see the values in the plugin screen. I haven't done any long term testing of it yet like reboot survival etc but I will. I do see the values updating so thats good. maybe it was permission issue?

vitormhenrique, 2017-03-27:
DHT sensors are very unstable on linux, because of the IO timing, DS18B20  are better but don't provide humidity. I'll search for alternative sensors and add to the plugin. Another option is to plug the temperature sensor on a small arduino mini clone, and have that communicate to the raspberry pi the temperature using I2C, this would be much more reliable.

closed 2017-03-27 by vitormhenrique

vitormhenrique, 2017-03-27:
I think a good candidate for temperature and humidity sensor is the https://www.adafruit.com/products/3251

Not super expensive, good and should be more reliable, maybe someday I'll get one and implement it. If someone ships any sensor for my I'll gladly test and add it as well.

mcmasterp, 2017-03-27:
where would you like one to arrive? I'll have it ordered today

vitormhenrique, 2017-03-27:
hey, relax, let me work on the initial improvements first....

vitormhenrique, 2017-03-27:
By the way, I don't have a link for the project of my arcade yet, I'll create one when done. 
If you really really want better temperature AND humidity I can give you the address, just send a email to vitormhenrique@gmail.com, if you just want temperature, just use the DS18B20  :)

mcmasterp, 2017-03-27:
haha no problem. My apologies, I only meant to give you the freedom to work on it when the whim struck you. You can of course take as long as you'd like to get around to, or even to change your mind and not do any of it and I would still be profoundly grateful for your work.  

I assumed you had a good use case for the humidity reading, I haven't thought of a good use so I guess i'll go ds18b20 then. in the future if it has a good use I'm all in

vitormhenrique, 2017-03-27:
No need to apologies! I just don't want to get something that I know it is going to be few weeks before I have the chance to work on now... 
I usually like to keep an eye on the humidity for high hygroscopic materials, So I can heat the enclosure and lower the humidity of the air inside and then print getting best performance and avoiding making the filament wet.

mcmasterp, 2017-03-27:
I ordered myself a DS18B20 for now. I only print in pla really so its more a "nice to have" for the humidity, but now you make me get ideas like peltier devices filament path to change temp and or humidy. 6 small peltiers, 3 in a row on either side of the filament would give you interesting abilities.

I had an email sent to you by adafruit, if it didn't arrive check spam.

vitormhenrique, 2017-03-27:
haha! I think peltier devices would be a little bit overkill ;)

I would love to see your co2 cartridge fire extinguisher, another project that I've been working on for a few days is a co2 cartridge powered nerf gun. I did the initial testing and it is going to be awesome, be careful with the solenoid that you use, as co2 cartridges contain high pressure, and might be dangerous. I'm using a regulator between my solenoid and using a high quality 350 psi rated solenoid valve, but the regulator and the solenoid were a little bit expensive.

I just got the email, thank you very much, I appreciated, but you know that it was definitely not needed right? my goal is too help people out with this plugin, I started this project to fill gaps that I had on my printer and shared because I thought that people could beneficiate, never the less I really appreciate the gesture :)

Shoot me a email and we can share some projects and ideas..

mcmasterp, 2017-03-27:
Oh my I hadn't even thought of regulating the pressure before the solenoid. I may opt for a mini co2 regulator meant for beer kegs. Either that or an expansion tank to lower the pressure

Oh I know it wasn't needed, i can tell you do it for other than money. My goal is to help enable people what little ways i can. It seems a fair deal considering. 

I'll email you