vitormhenrique/OctoPrint-Enclosure #15: MOAR IO!

mcmasterp, 2017-03-27:
Bet you saw this coming. I need one more io at least lol. I ended up putting hotend, ramps, and an eclosure fan on relays, plus the lights and I'm maxed out. I holding out hope for temp triggers some point down the line so I am installing the fire supression soleniod setup now since I have the parts ready.

vitormhenrique, 2017-03-27:
I'm already coding the new plugin, spoiler alert: temperature triggers, and as many io as you want..  basically I'm changing the entire software architecture and allowing creating as many inputs and outputs as you want...

It is going to take few more days because I'm also working on my arcade machine.

mcmasterp, 2017-03-27:
NICE! thank from everyone.

also, did you say arcade???? my best maker build to date is a bartop for my father in law
[https://hackaday.io/project/13069-murphcade](url)

Got a link to the project?

vitormhenrique, 2017-04-11:
Implemented on version 3.0 that is on dev branch right now, if you want to test it out uninstall the current plugin and install it again manually using the development branch. I still need to do more testing, all architecture of the software changed, so it might have few bugs.
 https://github.com/vitormhenrique/OctoPrint-Enclosure/archive/dev.zip

closed 2017-04-14 by vitormhenrique