vitormhenrique/OctoPrint-Enclosure #16: RuntimeError:  Failed to add edge detection

trgtylcnky, 2017-03-27:
Hi,
Filament sensor won't work for me. I see  "RuntimeError: Failed to add edge detection" in logs.
I think it is about permissions. I added pi user to gpio group and my system is up to date.

vitormhenrique, 2017-03-27:
Are you using any other plugin that might use GPIO?

vitormhenrique, 2017-03-27:
Another thing that it is likely is that you are using the wrong pin numbers, by default the plugin uses BCM pin numbering, they are different from BOARD numbers.

For example BOARD pin number 11 is BCM pin number 17

[image]

trgtylcnky, 2017-03-29:
It seems it is a kernel problem. When I look at the files:
`
ls /sys/class/gpio/gpio14/ -l
total 0
-rw-r--r-- 1 root root 4096 Mar 29 14:28 active_low
lrwxrwxrwx 1 root root    0 Mar 29 14:28 device -> ../../../gpiochip0
-rw-r--r-- 1 root root 4096 Mar 29 14:28 direction
-rw-r--r-- 1 root root 4096 Mar 29 14:28 edge
drwxr-xr-x 2 root root    0 Mar 29 14:28 power
lrwxrwxrwx 1 root root    0 Mar 29 14:28 subsystem -> ../../../../../../../class/gpio
-rw-r--r-- 1 root root 4096 Mar 29 14:28 uevent
-rw-r--r-- 1 root root 4096 Mar 29 14:28 value
`
Since it is not related to your plugin, you can close the issue. But I wonder which linux kernel version you are using.

vitormhenrique, 2017-03-29:
I have 4.1.19-v7+, download the latest octopi image, and it should work fine.

closed 2017-03-29 by vitormhenrique

trgtylcnky, 2017-03-29:
I am using 4.9.17 since I upgraded the kernel to be able to run chromium for TouchUI. Probably it had broken on new kernel.

vitormhenrique, 2017-03-29:
Chromium worked ok for me on 4.1.19 with the raspberry pi touchscreen lcd. Used for a while, but them decided that it is easier, at least for me, having bookmarks on my home screen and use octoprint with my cellphone.

ziesemer, 2017-05-12:
See also:  https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=183481

vitormhenrique, 2017-05-12:
thanks @ziesemer for more information on this issue!