vitormhenrique/OctoPrint-Enclosure #17: LM35 Temp Sensor

soylentgeek, 2017-04-03:
Could you add support for an LM35 temp sensor?

I have one wired up already thinking it might work, but unlike a DS18B20 it creates a 00-xxxxxxx directory and not a 28-xxxxxxx

Thanks,

D.

vitormhenrique, 2017-04-03:
Are you sure you have the correct sensor name?

Doing a research I found that [LM35](http://www.ti.com/product/LM35) has an analog output, instead of a one wire interface.

Can you point me to the correct datasheet?

soylentgeek, 2017-04-03:
This is the device I have.

https://www.osepp.com/electronic-modules/sensor-modules/79-lm35-temperature-sensor-module

Thanks

vitormhenrique, 2017-04-03:
That is definitely the same sensor that I sent you, it is not a one wire sensor. Have you tested on your raspberry pi? do you have the code that you are using? You can see on the page that you sent me that the sensor is connected to the analog zero on the arduino, the temperature read will create a proportional voltage on the output pin....

Raspberry Pi don't have an ADC, so you would need an MCP3008 or something similar to read the analog voltage from the sensor and convert to temperature.

soylentgeek, 2017-04-03:
Thanks for the feedback,

I will just try and find the already supported sensors in this case...
sounding too complicated to do it using what I have.

Thanks for taking the time to help.

D.

vitormhenrique, 2017-04-03:
no problem :)

closed 2017-04-03 by vitormhenrique

Sleepybear, 2020-12-01:
Hi!

I know this is a closed issue, but I was hoping you could help. 

I had some LM35 sensors, and for fun I got an MCP3008 ADC and hooked it up to the PI. I'm able to read in temp numbers from a python script using adafruit MCP3xxx library. How can I add this sensor to Enclosure Plugin?

It uses the SPI, and I put it on CE0, the only other sensor I could find that uses that is the MAX 31855. Is there a way that I can trick it to seeing the readings?