vitormhenrique/OctoPrint-Enclosure #19: Default temperature for enclosure

trgtylcnky, 2017-04-06:
Hi,
It would be nice to have a default temperature value in settings. It starts heating automatically when octoprint starts.

sajovicd, 2017-04-06:
I dont think that would be a good idea To start heating when you turn on octopi. Or you mean whe you start printing?

vitormhenrique, 2017-04-06:
I think the meant when started printing.

trgtylcnky, 2017-04-06:
Yes, you are right.  I updated

vitormhenrique, 2017-04-11:
Implemented on version 3.0 that is on dev branch right now, if you want to test it out uninstall the current plugin and install it again manually using the development branch. I still need to do more testing, all architecture of the software changed, so it might have few bugs.
 https://github.com/vitormhenrique/OctoPrint-Enclosure/archive/dev.zip

vitormhenrique, 2017-04-14:
Default temperature is not implemented, enclosure temperature control will start automatically  when the print starts.

closed 2017-04-14 by vitormhenrique