vitormhenrique/OctoPrint-Enclosure #21: serial connection to marlin 1.1 fails when toggling gpios

mcmasterp, 2017-04-11:
I setup five gpio outputs and went tot  test them. if i toggle them for a bit at some point the printer disconnects with error:

Offline: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:1792

debug was enabled, here's log:

2017-04-11 20:05:26,089 - octoprint.plugins.enclosure - INFO - temperature_reading: [{u'sensorType': u'18b20', u'isEnabled': True, u'gpioPin': 4, u'useFahrenheit': False}]
2017-04-11 20:05:26,091 - octoprint.plugins.enclosure - INFO - temperature_control: [{u'autoStartup': False, u'isEnabled': False, u'defaultTemp': u'40', u'controlType': u'cooler', u'activeLow': True, u'gpioPin': u'3'}]
2017-04-11 20:05:26,092 - octoprint.plugins.enclosure - INFO - rpi_outputs: [{u'gpioPin': u'17', u'autoShutdown': True, u'label': u'Lights', u'autoStartup': True, u'active': False, u'activeLow': True, u'shutdownTimeDelay': u'3600', u'startupTimeDelay': 0}, {u'autoShutdown': True, u'shutdownTimeDelay': u'180', u'autoStartup': True, u'startupTimeDelay': 0, u'active': False, u'activeLow': True, u'label': u'Ramps Fan', u'gpioPin': u'27'}, {u'autoShutdown': False, u'shutdownTimeDelay': 0, u'autoStartup': False, u'startupTimeDelay': 0, u'active': False, u'activeLow': True, u'label': u'Fire Extinguisher', u'gpioPin': u'18'}, {u'autoShutdown': True, u'shutdownTimeDelay': u'3600', u'autoStartup': True, u'startupTimeDelay': 0, u'active': False, u'activeLow': True, u'label': u'Hotend Fan', u'gpioPin': u'22'}, {u'autoShutdown': False, u'shutdownTimeDelay': 0, u'autoStartup': False, u'startupTimeDelay': 0, u'active': False, u'activeLow': True, u'label': u'Cooling', u'gpioPin': u'3'}]
2017-04-11 20:05:26,093 - octoprint.plugins.enclosure - INFO - rpi_inputs: [{u'setTemp': 100, u'eventType': u'temperature', u'controlledIO': u'17', u'label': u'', u'inputPull': u'inputPullUp', u'edge': u'fall', u'setControlledIO': u'low', u'gpioPin': 0}]
2017-04-11 20:05:29,978 - octoprint.plugins.enclosure - INFO - Temperature read was: 25.4
2017-04-11 20:05:33,404 - octoprint.util.comm - ERROR - Unexpected error while reading from serial port
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.2-py2.7.egg/octoprint/util/comm.py", line 1792, in _readline
    ret = self._serial.readline()
  File "build/bdist.linux-armv7l/egg/serial/serialposix.py", line 475, in read
    raise SerialException('device reports readiness to read but returned no data (device disconnected or multiple access on port?)')
SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2017-04-11 20:05:40,859 - octoprint.plugins.enclosure - INFO - Temperature read was: 25.4
2017-04-11 20:05:51,708 - octoprint.plugins.enclosure - INFO - Temperature read was: 25.4
2017-04-11 20:05:56,375 - octoprint.util.comm - INFO - Printer reports firmware name "Marlin 1.1.0-RC8 (Github)"
2017-04-11 20:05:58,525 - octoprint.util.comm - INFO - Printer reports firmware name "Marlin 1.1.0-RC8 (Github)"
2017-04-11 20:06:02,578 - octoprint.plugins.enclosure - INFO - Temperature read was: 25.4
2017-04-11 20:06:13,449 - octoprint.plugins.enclosure - INFO - Temperature read was: 25.4
2017-04-11 20:06:24,319 - octoprint.plugins.enclosure - INFO - Temperature read was: 25.5
2017-04-11 20:06:34,084 - octoprint.util.comm - ERROR - Unexpected error while reading from serial port
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.2-py2.7.egg/octoprint/util/comm.py", line 1792, in _readline
    ret = self._serial.readline()
  File "build/bdist.linux-armv7l/egg/serial/serialposix.py", line 475, in read
    raise SerialException('device reports readiness to read but returned no data (device disconnected or multiple access on port?)')
SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2017-04-11 20:06:35,188 - octoprint.plugins.enclosure - INFO - Temperature read was: 25.5
2017-04-11 20:06:46,058 - octoprint.plugins.enclosure - INFO - Temperature read was: 25.5
2017-04-11 20:06:56,909 - octoprint.plugins.enclosure - INFO - Temperature read was: 25.5
2017-04-11 20:07:05,832 - octoprint.server.util.sockjs - INFO - Client connection closed: 192.168.1.17

vitormhenrique, 2017-04-11:
the error came from octoprint, not the plugin, try to restart the server....

mcmasterp, 2017-04-11:
restart didn't help but I have it narrowed down to the gpio controlling a single relay board that is controlling a 12v solenoid for fire suppression ideas. It seems that when I turn that device off, it causes the issue. it could be rf interference? power surge? or of course bad relay, which I will switch asap

vitormhenrique, 2017-04-11:
All those are good ideas, avoid having the usb cable near the solenoid, or it's cable. Also if possible use different power supply, that should fix your issue.

closed 2017-04-11 by vitormhenrique