vitormhenrique/OctoPrint-Enclosure #26: temperature broken with new upate

elfhelmp, 2017-04-15:
Hello,
I updated your plugin to the latest version and after reboot I couldn't see anymore the temp and humidity readings, it was just blank. I tried reinstalling, nothing. I wiped the SD card and started with octoprint and the plugin from scratch, still it won't show any values. I looked for the folders /.octoprint/plugins/octoprint-enclosure/extras/ but they are all missing now, they haven't been created when the plugin was installed via Plugin Manager. I no longer find GetHumidity.py and GetTemperature.py scripts in there, are they no longer required? I tried to add everything but I can no longer locate them at "https://raw.githubusercontent.com/vitormhenrique/OctoPrint-Enclosure/master/extras/GetHumidity.py ".
Please help me figure this out cause I've been struggling to get it to work but nothing. If I run the example scripts to check the sensors, it outputs proper values.

elfhelmp, 2017-04-15:
Update: log can be found here: https://pastebin.com/iM4T5XG7
using DHT11 sensor with Octoprint 1.3.2

Error I get is this: 
2017-04-15 02:20:59,398 - octoprint.plugins.enclosure - INFO - sensorType dht
2017-04-15 02:20:59,411 - octoprint.plugins.enclosure - WARNING - An exception of type AttributeError occurred on checkEnclosureTemp. Arguments:
("'RepeatedTimer' object has no attribute 'stop'",)

vitormhenrique, 2017-04-15:
can you try again?
uninstall and install it again....

vitormhenrique, 2017-04-15:
scripts are no longer required... but I did not test the dht sensor.... just ported the code...

elfhelmp, 2017-04-15:
I reinstalled everything from scratch. I put both Octoprint v1.3.2 and 1.3.0 and 0.14.0 Octopi. I followed the instructions step by step and now I have a new error, it says "Failed to import Adafruit_DHT. Please install library and restart Octoprint." I installed the library and I did local test, DHT sensor works.

vitormhenrique, 2017-04-15:
I just figure that octoprint does not have wrists to access the GPIO, I was trouble shooting this until now... 

I'll have to fix this tomorrow, 4 am here =/

vitormhenrique, 2017-04-15:
5am got the temperature working, tomorrow I'll fix the humidity

elfhelmp, 2017-04-15:
Haha many thanks man but it's not that urgent. I pulled an all-night as well troubleshooting this but next week is also ok. I just wonder if this happens to all who update or just on my RPI? I will check later to see if temp works.

vitormhenrique, 2017-04-15:
got it working now, I tried to make the installation of the adafruit library automatically and that broke the plugin before, Can you instal 3.4 and give it a try?

elfhelmp, 2017-04-15:
Still not ok mate, I'm sorry. On the Enclosure plugin the temp and humidity have the same value of 25 while on test SSH I have 25 temp and 62 humidity. I think it reads only temp and displays it in both values. Many thanks for the quick responses so far!

vitormhenrique, 2017-04-15:
can you uninstall, install it again using the plugin manager, from url and put https://github.com/vitormhenrique/OctoPrint-Enclosure/archive/master.zip

Also, make sure to enable debug on the advanced.

I aded more log information to help us troubleshoot.

vitormhenrique, 2017-04-15:
I'm an idiot! I forgot to push the updated script! 
3.42 should have it...

NiklasOlofsson, 2017-04-15:
I use the AM2302-sensor.
```
pi@octopi:~/Adafruit_Python_DHT/examples $ ./AdafruitDHT.py 2302 4
Temp=22.9*  Humidity=38.7%
```

The plugin shows temp 0, hum 0:

```
2017-04-15 22:44:27,383 - octoprint.plugins.enclosure - INFO - sensorType dht
2017-04-15 22:44:27,387 - octoprint.plugins.enclosure - INFO - Temperature dht cmd: sudo python /home/pi/oprint/lib/python2.7/site-packages/octoprint_enclosure/getDHTTemp.py 2303 4
2017-04-15 22:44:27,676 - octoprint.plugins.enclosure - INFO - Temperature dht result: 
2017-04-15 22:44:27,677 - octoprint.plugins.enclosure - WARNING - An exception of type ValueError occurred on readDhtTemp. Arguments:
('need more than 1 value to unpack',)
2017-04-15 22:44:27,679 - octoprint.plugins.enclosure - INFO - Temperature: 0 humidity 0
```

vitormhenrique, 2017-04-15:
and if you type "sudo python /home/pi/oprint/lib/python2.7/site-packages/octoprint_enclosure/getDHTTemp.py 2303 4" 

what happens?

NiklasOlofsson, 2017-04-15:
Nothing, I guess? ;)

```
pi@octopi:~/Adafruit_Python_DHT/examples $ sudo python /home/pi/oprint/lib/python2.7/site-packages/octoprint_enclosure/getDHTTemp.py 2303 4
pi@octopi:~/Adafruit_Python_DHT/examples $
```

vitormhenrique, 2017-04-15:
can you see if the file exists? and it's content? 

cat /home/pi/oprint/lib/python2.7/site-packages/octoprint_enclosure/getDHTTemp.py

NiklasOlofsson, 2017-04-15:
Yes, the file exists. The command would tell me file not found if not.

```
import sys
import Adafruit_DHT

# Parse command line parameters.
sensor_args = { '11': Adafruit_DHT.DHT11,
    '22': Adafruit_DHT.DHT22,
        '2302': Adafruit_DHT.AM2302 }
if len(sys.argv) == 3 and sys.argv[1] in sensor_args:
    sensor = sensor_args[sys.argv[1]]
    pin = sys.argv[2]
else:
    sys.exit(1)

humidity, temperature = Adafruit_DHT.read_retry(sensor, pin,2,0.5)

if humidity is not None and temperature is not None:
    print('{0:0.1f} | {1:0.1f}'.format(temperature, humidity))
else:
    print('-1 | -1')

sys.exit(1)

```

vitormhenrique, 2017-04-15:
I was passing 2303 instead of 2302...

select the 22 sensor for now, it will work the same...  I'll fix this....

NiklasOlofsson, 2017-04-15:
Yes. 22 works. Thanks!

vitormhenrique, 2017-04-15:
should be fixed on 3.43 as well...

closed 2017-04-15 by vitormhenrique

elfhelmp, 2017-04-16:
All is working now for me as well