vitormhenrique/OctoPrint-Enclosure #27: Output state changes on config change

rtheil, 2017-04-15:
If I make any changes to enclosure config, outputs always turn off rather than staying in their previous state.

Reproduce:
1. Create output and turn output on/high
2. Change any enclosure config item (does not have to be related to output)
3. Save config. Output is now in low/off state.

In my case, it turns off the printer.

Additionally, when octoprint server is restarted, outputs return to low/off state.

vitormhenrique, 2017-04-15:
the thing is that it needs to reconfigure after you save your settings... 
But I think I can bypass the used pins that did not change... 

I'll do some testing

vitormhenrique, 2017-04-15:
one question, if your output turns on on high, are you selecting deselecting the activeLow  checkbox?

rtheil, 2017-04-16:
The activeLow checkbox is NOT checked.

Previous version always left outputs alone when reconfiguring or restarting octoprint.

vitormhenrique, 2017-04-18:
Fixed on 3.45

closed 2017-04-18 by vitormhenrique

rtheil, 2017-04-27:
When restarting OctoPrint server, the outputs are reset and the output button state shows that it's in the wrong position. For example, I turn on printer output and fan output. Then I restart OctoPrint server. After restart, enclosure plugin shows that fan and printer are on, but they are actually off.

The only reason this is a problem is because I use the GPX plugin for my printer. If I have to cancel a print, I MUST restart Octoprint or when I restart the print it will sometimes stop after the first few layers because the GPX plugin is still stuck on the previous operation. Known issue with GPX plugin. It's the only reason why it'd be nice if the output state didn't change on OctoPrint restart.

vitormhenrique, 2017-04-30:
I think I fixed this on the next build.

rtheil, 2017-04-30:
Awesome, thanks! I will test it out and report back.

rtheil, 2017-05-24:
Looks like this one isn't quite fixed. If I turn on an output, then restart octoprint server, output is then turned off. Difference from before is that the buttons now show the proper state that the output is actually in.