vitormhenrique/OctoPrint-Enclosure #28: Temperature and humidity in the navbar?

NiklasOlofsson, 2017-04-16:
Hello,

Could the temperature and humidity be shown like this:
[image]

Maybe the possibility to set the names for the two values yourself (to make it fit)? Like T: 22C H: 35%

vitormhenrique, 2017-04-18:
I'll double check to see if adding a new item to the navbar works together with the "Navbar Temperature Plugin", if not I'll re-implement everything on my plugin....
I'll only be able to check this the following weekend....

vitormhenrique, 2017-05-12:
added on 3.5

closed 2017-05-12 by vitormhenrique

NiklasOlofsson, 2017-05-12:
Is it possible to turn it off if you need to?

vitormhenrique, 2017-05-12:
Not yet, but i can add it on next release....

Best regards,
Vitor Henrique

vitormhenrique, 2017-05-12:
3.51 you can turn it off on advanced settings....