vitormhenrique/OctoPrint-Enclosure #30: Temperature sensor shown values

rtheil, 2017-04-17:
Just got a DHT11 connected and am testing that out. Have a suggestion and issues.

1. Please show temperature type indicator (F/C) next to temperature readout

2. The temperature shown sometimes has massive precision. I saw a temp a minute ago that showed something like (69.8000000000). Couldn't see beyond the last 0, as it overfilled the table cell.

3. Temperature readout is all over the place sometimes. Right now it's about 68 degrees, and usually shows 68, but sometimes shows completely random numbers like 154.4 and 300+. I assume this is a return from the DHT library, but I'm unsure how I could test this further at a command prompt.

rtheil, 2017-04-17:
Was able to copy the full temp with extreme precision: "69.80000000000001	"

vitormhenrique, 2017-04-17:
1) I'm showing the unit only on the target area, if temperature control is enable, because I was using the same layout of the printer temperature control, I'll make it appear on the "reading" area as well....

2) Easy, I'll take care of that...

3) I think I know what is causing the big values, Don't even need to see the code, I think there is a bug...

rtheil, 2017-04-17:
Awesome! In my case I am not using the target yet, as I don't have a relay for a heater yet. But extra relays are on their way from china, so that's coming soon.

Would it be ridiculously difficult to add support for the BME280 temperature sensor? I tested the DHT11/22 and others awhile back on a prototype I was working on, and found the DHT to be really imprecise (because analog), though it's probably fine for this setup. The BME280 ended up becoming my best buddy, and is highly accurate. Big difference is that it requires SDA/SCL.

rtheil, 2017-04-17:
Also, maybe add % symbol next to humidity while you're in there?

vitormhenrique, 2017-04-17:
I'm planing to add the Si7021, bought one unit and it is coming on the mail this week. it is inexpensive and should have more than enough precision. I don't have the BME280 to test, if you can give me a python script that reads the temperature and the instructions to hook it up on the raspberry pi i can implement it and leave the testing to you... if not we will have to wait me getting my hands on this sensor....
I hate the DHT sensors on the rpi, not because the imprecision, but if you start noticing they will consume a lot of cpu on failed attempts...  for arduino stuff they are good. I'm planing to stick with Si7021 on my printers.

rtheil, 2017-04-17:
Interesting... Unfortunately, my Python dev knowledge is non-existent, however I will look around for something that already exists. What's the required output of the python script? Just something parseable, like "00|11" where 00=temperature and 11=humidity? BME280 also puts out barometric pressure, but that's probably not useful for this implementation.

rtheil, 2017-04-17:
Looks like Adafruit already has a python lib for BME280. Sweet!
https://github.com/adafruit/Adafruit_Python_BME280

vitormhenrique, 2017-04-18:
Fixed all issues on this thread on 3.45, please open new issue for BME280 support. :)
Thank you for testing...

closed 2017-04-18 by vitormhenrique