vitormhenrique/OctoPrint-Enclosure #31: BME280 Temp/Humidity sensor support

rtheil, 2017-04-18:
Adafruit already has a python lib for BME280.
https://github.com/adafruit/Adafruit_Python_BME280

Happy to test this out when you're ready.

elfhelmp, 2017-04-27:
I have one at the ready as well, if you ever implement this let me know and I'll put it to good use.

rtheil, 2017-04-28:
Victor, I'm willing to make a donation to you for all your hard work on this project, and so you can purchase a BME280 for easier development and implementation of its support.

mcmasterp, 2017-04-28:
That's a really good idea. I had the same thought for other features and generally as a thank you and to help fund the next project he hasn't even thought of yet. I highly encourage everyone using and especially those interacting here to drop a few bucks on some gift cards to the usual places. 

Sorry I'm not affiliated just passionate about supporting open source freeware development

sajovicd, 2017-04-28:
I am also willing to contribut.
This is realy great men and willing to help.
All respect.

mcmasterp, 2017-04-28:
Nice! My new slogan is 

"give till it hurts when your wife hits you for it!" 

Or husband of course but I got a wife and she just got a whole new pack of wooden spoons

vitormhenrique, 2017-04-28:
haha... Thanks guys. I really appreciate the gesture, my biggest constrain today is TIME, to many projects, and a day job that take all my time and energy. Firstly I would suggest donating to octoprint itself, despite the hours that I put on this plugin they will never compare to what @foosel did to make octoprint available to us.

Of course I can not buy every single sensor that exist and test them, but I'll make my best and leave you guys to the testing.

If you really, really want to donate something, just send me an email and we will talk about it...

I'll try implementing this sensor and another, that I got from @mcmasterp contribution this weekend.

mcmasterp, 2017-04-28:
+1 @foosel donations

vitormhenrique, 2017-04-30:
What's up guys, 

Before I push my changes could someone test my script and instructions to get the values from rpi?

1. Run sudo raspi-config
2. Use the down arrow to select 9 Advanced Options
3. Arrow down to A7 I2C
4. Select yes when it asks you to enable I2C
5. Also select yes when it asks about automatically loading the kernel module
6. Use the right arrow to select the <Finish> button
7. Select yes when it asks to reboot

8.Install some packages:
`sudo apt-get install i2c-tools python-pip` 

9.Install the driver:
`sudo pip install RPi.bme280`

10.Create a python script:
`nano bme280.py`

11.Paste the following code
<pre><code>
import smbus2
import bme280

port = 1
address = 0x77
bus = smbus2.SMBus(port)

bme280.load_calibration_params(bus, address)

# the sample method will take a single reading and return a
# compensated_reading object
data = bme280.sample(bus, address)

# the compensated_reading class has the following attributes
print(data.temperature)
print(data.humidity)
</code></pre>

11.Run the python script:
`python bme280.py`

vitormhenrique, 2017-04-30:
of course you need to connect the sensor:
VIN              to	3V3
GND	    to	        GND
SCL	Clock   to	GPIO 3 (SCL)
SDA	Data     to	GPIO 2 (SDA)

rtheil, 2017-04-30:
#6 should be: "Use the right arrow to select Finish"

What directory should bme280.py be in?

I will try #11 tomorrow when I can connect the BME280.

vitormhenrique, 2017-04-30:
@rtheil it does not matter the location, this is just to test the python script...

rtheil, 2017-05-01:
Just ran that script and got this:

Traceback (most recent call last):
  File "bme280.py", line 2, in <module>
    import bme280
  File "/home/pi/bme280.py", line 8, in <module>
    bme280.load_calibration_params(bus, address)
AttributeError: 'module' object has no attribute 'load_calibration_params'

I verified that pip has the bme280 driver installed

sudo pip install RPi.bme280
Requirement already satisfied (use --upgrade to upgrade): RPi.bme280 in /usr/local/lib/python2.7/dist-packages
Requirement already satisfied (use --upgrade to upgrade): smbus2 in /usr/local/lib/python2.7/dist-packages (from RPi.bme280)
Cleaning up...

vitormhenrique, 2017-05-01:
weird!

the driver definitely has the function defined:

https://github.com/rm-hull/bme280/blob/master/bme280/__init__.py#L118

I'll take a look later today when I go back home....

rtheil, 2017-05-01:
I will try and dig deeper, but I know literally zero about Python, even though I'm a seasoned C# developer.

Let me know if it would help any if I gave you SSH access to my Pi. I could poke a hole in the firewall for you, as long as you can give me the IP address you'll be connecting from.

vitormhenrique, 2017-05-02:
@rtheil Can you send me an email (mine is on my github page info)

on my raspberry pi, I dont get the same error that you are having... I might need to do on a raspberry pi with the sensor plugged in...

vitormhenrique, 2017-05-06:
Hello everyone, I'm finishing the new release with a bunch of improvements and new features...
I added the si7021, and i2c sensor that @mcmasterp donated to me. I tested and got it working. 
I can't add the BME280 if anybody test the script, let me know the issues and we can figure it out. 
Ideally give me and SSH of the raspberry pi that has the sensor so I can play with it.
If I can't have this working by Monday I'll publish the next release without the sensor and it will only be included on the next release.

vitormhenrique, 2017-05-08:
bueller bueller bueller.... @rtheil @elfhelmp @sajovicd

sajovicd, 2017-05-08:
?

vitormhenrique, 2017-05-08:
@sajovicd I'm trying to test the sensor for a while now and got no answer....  
[bueller](http://www.urbandictionary.com/define.php?term=Bueller)

sajovicd, 2017-05-08:
Ooo I get it 😂🤣
I watch youtube just now.

Well sorry I cant hrlp you with this sensor, I dont have it.

vitormhenrique, 2017-05-08:
@sajovicd no problem!

elfhelmp, 2017-05-08:
Tomorrow morning I will give it a shot and I'll sort a SSH to the raspberry as well with a BME280 attached. I will PM you with login details ASAP. (Hoping they won't call me tonight, I'm On-call this week).

vitormhenrique, 2017-05-09:
Got the script working thanks @elfhelmp. 
@rtheil were you using an raspberry pi zero model when trying the script?

rtheil, 2017-05-09:
Sorry for my lack of responses on this one. I've had some seriously busy days at work, and of course that's where my 3D printer is.

I am using a regular old Raspberry Pi 3. Do you have a new version of the script I can try out?

vitormhenrique, 2017-05-09:
@rtheil you can try this one... 
https://gist.github.com/vitormhenrique/2c4617353992666359b5493c1d684321

download on your raspberry pi, and run using
`python [NAME OF THE FILE] [address of the sensor]`

You can find the i2d address by running 
`i2cdetect -y 1`

It should display something like:
<pre><code>
       0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
  00:          -- -- -- -- -- -- -- -- -- -- -- -- --
  10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
  20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
  30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
  40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
  50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
  60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
  70: -- -- -- -- -- -- 76 -- </code></pre>

vitormhenrique, 2017-05-12:
added on 3.5

closed 2017-05-12 by vitormhenrique

Ricardosgeral, 2018-06-20:
hi 
I have

pi@raspberrypi:~/relier $ python3 bme280.py 76
Traceback (most recent call last):
  File "bme280.py", line 2, in <module>
    import bme280
  File "/home/pi/relier/bme280.py", line 8, in <module>
    calibration_params = bme280.load_calibration_params(bus, address)
AttributeError: module 'bme280' has no attribute 'load_calibration_params'

can figure it out

vitormhenrique, 2018-06-20:
Hello, I don't think you are using the same file that I have on my github, the file here does not have anything related with calibration params.

https://raw.githubusercontent.com/vitormhenrique/OctoPrint-Enclosure/master/octoprint_enclosure/BME280.py

Are you trying to setup the temperature sensor with octopi? or just found this repository randomly?

Ricardosgeral, 2018-06-20:
The problem is that I had a bme280.py file... Had to change that name.
Doing import bme280 called that file dahhh