vitormhenrique/OctoPrint-Enclosure #32: PWM Outputs

vitormhenrique, 2017-04-18:
Enable PWM output

DevKitchen, 2017-04-25:
I think PWM is a hardware feature that Pi doesn't have, you'd need an external controller to make it work. Arduinos have PWM features though.

mcmasterp, 2017-04-25:
Good question. I'm surprised it doesn't have pwm on a few pins. 

Here's a fevered dream: the plugin has a setting where you input the ip address or host name of a esp8266 that is running special client code. then you can not o lying have pwm but also more sensor support and of course the fact that the pi and esp8266 are untethered from each other makes tons of possibilities! 

There's really so much potential for this plugin to become a must have for every octopi when you think about it. You don't even need an enclosure to want it!

flyer007, 2017-04-25:
Pi has both software PWM and hardware PWM
https://raspberrypi.stackexchange.com/questions/298/can-i-use-the-gpio-for-pulse-width-modulation-pwm

Codsworthy, 2017-04-29:
Oh awesome! I did some rudimentary looking around as I bought 2 PWM fans without thinking about it and found nothing on Pi PWM only Arduino. I'll be using this info I think!

vitormhenrique, 2017-04-29:
as @flyer007 said RPI does have PWM support, and I'm using RPi.GPIO python library to control IO's so I'll be using software PWM.

@Codsworthy you don't need an PWM output to control a fan, PWM will only control the speed of the fan, so if you give it a maximum voltage, 5v on it's output the fan will run at full power. If you want to slow the fan, you can create a voltage divider using two resistors and use that pin to control the fan speed.

@mcmasterp I'm working on creating wireless filament sensors, I hate wires dangling around my print area,  but I'll probably use nRF24LU1 transceivers on my final design.

mcmasterp, 2017-04-29:
Cool wireless support for io in general has lots of potential. I mounted my sensor in the ceiling of the enclosure so the wiring is run along ceiling then down wall to where pi is mounted. I'll post link to some pics when I get home

vitormhenrique, 2017-05-12:
added on 3.5. 
Note that 0% duty cycle is 100% on... and so on.....

closed 2017-05-12 by vitormhenrique