vitormhenrique/OctoPrint-Enclosure #33: GPIO only works after toggling "Use Board Pin".

traveltje2, 2017-04-19:
I have a led connected to GPIO17.
Every time I reboot octoprint I need to toggle the "Use Board Pin #", save it and toggle it again before the GPIO works. After the toggle everything works as expected until a reboot.

vitormhenrique, 2017-04-19:
Ok, I'm a little confused, are you using board pin numbers or bcm? 

By default, this plugin will use BCM pin numbers, if you have if configured any time, for example, even open and saving anything with BCM configured, you won't be able to switch between the BCM and BOARD.

That is why there is a button to "clear gpio mode".

The correct workflow to change between modes is go to the setting screen, clear the GPIO mode, select whatever you want, and save the setting. If you do this, the next time you reboot, the plugin should keep the BOARD number checked and configure it as so....

have you tried this work flow?

traveltje2, 2017-04-19:
Hi, thanks for the fast reply.
I use the normal GPIO pin nr's (tried different pins with the same result.

Your workflow works perfect until a reboot. 
After the reboot I need to follow your workflow again to make it work. 
All other settings are stored and remembered after a reboot (for instance the GPIO17 and name I gave it).

vitormhenrique, 2017-04-19:
So the setting for the BOARD mode is not being remembered after the reboot right?
If so, I might know why, and will fix tonight...

traveltje2, 2017-04-20:
Thats my thought!

vitormhenrique, 2017-05-02:
@traveltje2  I could nor reproduce this issue, by any chance are you using another plugin or any other script on your pi that might be using GPIO? 
I'll add more login information on next release so we can trouble shoot this....

vitormhenrique, 2017-05-14:
@traveltje2 do you still have this issue? I'm going to close this because you did not respond for a while, and I could not reproduce the issue. If needed just reopen the issue and give me more information.

closed 2017-05-14 by vitormhenrique