vitormhenrique/OctoPrint-Enclosure #34: Temperature charting

DevKitchen, 2017-04-25:
I have no idea if it's possible but adding a line to the temperature chart for enclosure temperature would be so helpful to me. I'm trying to see if my enclosure heating system is effective and well balanced but that's impossible without some kind of record of temperatures over time.

I already use OctoPrint-TempsGraph to improve my chart, so maybe there's a way to connect to that?

vitormhenrique, 2017-04-25:
The only way to accomplish this, that I'm aware, is to overwrite the temp graph area with my own template and add the enclosure data. 
Despite is a nice thing to add currently I don't plan to implement this. It is a considerable effort and there are other priorities on my roadmap.
Having said that, is not that I'll never implement, but don't expect this on the following month or so...
Pushes are always welcome, if the author of OctoPrint-TempsGraph wants to add some sort of api that allow me to send temperature of the enclosure with a time stamp I could do the change on my end to send him the values.

mcmasterp, 2017-04-25:
@OctoPrint-TempsGraph

Is that how you tag this so it will show up on their project?

Codsworthy, 2017-04-28:
@1r0b1n0 is the writer of Octoprint-Tempsgraph. I dropped him an issue on his plug-in to see if you guys can work together to make both plug-ins even more awesome!

vitormhenrique, 2017-04-28:
Thanks, I'll take a look on what is possible.

skyhisi, 2017-07-12:
Just as a note, OctoPrint issue 321 (https://github.com/foosel/OctoPrint/issues/321) would allow this feature to be added, but as that issue has been open for over 3 years, it may not get fixed any time soon.

vitormhenrique, 2017-07-13:
Gina gave me another option on how to implement this, I'll try to contact tempsgraph author and see if we can work together to add this.

xd1936, 2018-03-06:
Could we somehow report the temperature probe as another extruder to Octoprint?

vitormhenrique, 2018-03-06:
no, as @skyhisi  commented, that area of octoprint is not extendable right now, i can not override any data or have access to it, Gina (or any other developer really) needs to update octoprint code to enable that.

calston, 2018-11-11:
A good start might be just adding the chart on the enclosure plugin screen, and then fix octoprint upstream to provide some plugin hook to add other temperature data sources.

Morcegolas, 2019-04-26:
Thanks for the plugin, for now I'm only using it to show my enclosure temp and humidity, I was also searching for some kind of chart and found this, my need is a chart where I can see the temperature evolution over the time of the print. Hope we can see it implemented in the plugin because Octoprint-Tempsgraph don't seem being helping us out.

sciaio, 2019-07-12:
Looking forward to it

tsayles, 2020-06-20:
In her April 2020 YouTube series on plug-in development, I think Gina mentioned some temperature graphing work going on. Going back it looks like Jim Neill (@jneilliii ?) suggested ..." graph the output in the temp graph using my changes that you merged for cutom temps"... 

Is anyone working to leverage these updates for the Enclosure plug-in?

tsayles, 2020-06-20:
> In her April 2020 YouTube series on plug-in development, I think Gina mentioned some temperature graphing work going on. Going back it looks like Jim Neill (@jneilliii ?) suggested ..." graph the [...]

I think this might be the basis for implementing this enhancement. https://github.com/jneilliii/OctoPrint-PlotlyTempGraph

If no one else is currently working on this, I'll take a whack at it and if successful, post a pull request.

pyarmak, 2020-09-24:
I have a proof of concept based on @tsayles idea. Just something I threw together quickly. I don't really have time to make it robust right now but maybe it'll help someone.
```
# -*- coding: utf-8 -*-
from __future__ import absolute_import, unicode_literals

import octoprint.plugin

class TempSniffer(octoprint.plugin.StartupPlugin):
    def __init__(self):
        self.current_enc_temp = 0.0
        self.set_enc_temp = 0.0

    def on_after_startup(self):
        self._logger.info("Temp Sniffer loaded!")

        # Register to listen for messages from other plugins
        self._plugin_manager.register_message_receiver(self.on_plugin_message)

    def on_plugin_message(self, plugin, data, permissions=None):
        if plugin == "enclosure":
            if 'sensor_data' in data:
                self.current_enc_temp = data['sensor_data'][0]['temperature']
            elif 'set_temperature' in data:
                self.set_enc_temp = data['set_temperature'][1]

    def inject_temps(self, comm, parsed_temps):
        parsed_temps[u'enclosure'] = (self.current_enc_temp, self.set_enc_temp)

        return parsed_temps

__plugin_name__ = "Temp Sniffer"
__plugin_version__ = "0.0.1"
__plugin_description__ = "A quick \"Hello World\" example plugin for OctoPrint"
__plugin_pythoncompat__ = ">=2.7,<4"
__plugin_implementation__ = TempSniffer()
__plugin_hooks__ = {
    "octoprint.comm.protocol.temperatures.received": __plugin_implementation__.inject_temps
}
```
[image]

jneilliii, 2020-09-24:
Fair warning, I have not developed that PlotlyTempGraph plugin all the way and it will end up consuming lots of memory if you keep the web interface open because it's not currently clipping the graphing data in respect to the time cutoff. I was planning on getting back around to that at some point, but just wanted to give you guys a heads up.

readysetawesome, 2021-01-30:
> I have a proof of concept based on @tsayles idea. Just something I threw together quickly. I don't really have time to make it robust right now but maybe it'll help someone.
> 
> ...
> 
> [image]

This worked great, thank you.

jneilliii, 2021-01-30:
Someone created a fork for the Enclosure Plugin, not sure where @Dak0r is with a PR, if at all, but I'll be releasing an update to PlotlyTempGraph plugin shortly that incorporates a lot of fixes that work with it.

https://github.com/Dak0r/OctoPrint-Enclosure-with-Custom-Graphs

Dak0r, 2021-01-31:
> 
> 
> Someone created a fork for the Enclosure Plugin, not sure where @Dak0r is with a PR, if at all, but I'll be releasing an update to PlotlyTempGraph plugin shortly that incorporates a lot of fixes that work with it.
> 
> https://github.com/Dak0r/OctoPrint-Enclosure-with-Custom-Graphs

Since we managed to fix all open issues with your Plotly plugin in the latest 0.1.0 release, I'd be happy to prepare a PR soon.

My last concern is, that I added new settings for temperature sensors which are saved in the config (they define if a sensor is shown in the graph or not). This seems to require users to remove all existing rpi inputs before the update and re-add them after the update. I'd really like to avoid this, but I currently don't know if there's an existing solution for this? Maybe something like a config-patcher that is run after the update?

I currently simply added the settings to 'self.settingsViewModel.settings.plugins.enclosure.rpi_inputs'
https://github.com/Dak0r/OctoPrint-Enclosure-with-Custom-Graphs/blob/889b4d07c2dece52b2195a624452513eb88b15e3/octoprint_enclosure/static/js/enclosure.js#L460

jneilliii, 2021-01-31:
there is a settings migration approach that you could hook into, by iterating the setting version number and then doing what you need to do to the settings as part of the upgrade process. I have done this in a couple of my plugins. Here's an example.

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/blob/dd3568665a182452c456a694a5ba2f00dc542d19/octoprint_tplinksmartplug/__init__.py#L228-L295

Dak0r, 2021-02-09:
Thanks @jneilliii 
I just had time to check on this (and the other link you sent me) and got it working!
You can now easily switch between my fork and the official enclosure version, without any conflicts:
https://github.com/Dak0r/OctoPrint-Enclosure-with-Custom-Graphs

I will make a new fork of enclosures "development" branch, integrate my changes and make a clean PR some time this week! 😄

readysetawesome, 2021-10-21:
@pintail120  if you're not comfortable adding to the source code, then this is not yet accessible to you.  It might be worth it to follow this thread, as it's purpose is to eventually make a plugin or settings available to you as an end-user.