vitormhenrique/OctoPrint-Enclosure #35: Fan control via temperature

mike-berriman, 2017-04-28:
Hi,

Awesome plugin to start with.  I was planning something along these lines when yours was released - saved me a whole bunch of work.

My current problem is that I'm using temperature control to control a small fan.  Set up to turn on a relay at x degrees, it works perfectly.  However, when the temperature drops below x degrees again, the relay does not turn off.

Am I doing something wrong?
Is this the way it's designed?

Cheers,
Mike.

vitormhenrique, 2017-04-28:
Are you using the cooling option on the temperature control? That should do what you want to do...

The "event" option is intended to work as you described, You could add two events, one on the rising and another on the falling edge to accomplish the same thing, but just use the temperature control with the cooling option. 

If you are already using the cooling option on the temperature control area let me know...

mike-berriman, 2017-04-30:
I am using cooling option.  I think the problem might be that I was just testing - I had no print running.  I think it might behave differently if there is a print running.

I'll let you know.

vitormhenrique, 2017-05-02:
have you had the chance to test this? I could not reproduce the issue, also other people had tested this feature with no problem?
please, also ensure that you have the latest version installed...

mike-berriman, 2017-05-02:
Sorry no - I will hopefully be able to give it a try tonight.

mike-berriman, 2017-05-03:
Given it a test now - works fine.  When testing it, I was setting a value, and then trying it when a print wasn't running - then it doesn't work.  But when a print is running, works like a charm.  Perfect!

One question, is there any way to configure an output to a particular state when the system is first turned on?  On mine, it seems the fan (set up as both the temperature and a manual output) turns on, and the lights (output only) is turned off.  I'd like this to be reversed when turning the Pi on.  I've created a few event scripts which do it perfectly - until the enclosure plugin starts and turns them off again...

Thanks again - this is a great plugin

vitormhenrique, 2017-05-03:
Quick question, are you configuring the fan both on the temperature control and the output? You should only add it to one configuration.

When the raspberry pi is turned on, all outputs should be off, if they are not, you had something configured incorrectly. outputs will turn on automatically after the print starts if you configure it to do so....

closed 2017-05-03 by vitormhenrique

mike-berriman, 2017-05-03:
First, I configured the fan as temperature control only.  When I turned on the Pi, the fan started, and I had no control over it.  So I added a specific output for the fan control so that I had access to on and off buttons for manual control.  This seems to work absolutely perfectly.  The buttons even change correctly when the fan turns itself on due to temperature.

However, when restarting the Pi, the fan still turns on, so I have to manually turn it off.  I assume this is because of the temperature control, because the fan and light outputs are configured identically, and the lights are always off on startup.

So I think things configured as an output do turn off on startup.  This is the behaviour I am suggesting you should change.  It should be user configurable whether each out is on or off on startup.  That way I can turn my lights on as soon as the Pi starts, which is of course needed to set up for a print.

Thanks!

vitormhenrique, 2017-05-03:
Yeah, there is a bug with the cooling turning on because it thinks temperature control is set to 0 when turning on. Just change the default temperature control to something else, this will be fixed on the next release.

mike-berriman, 2017-05-03:
The default is already set to 40, and it still happens?  Doesn't matter if you have it in hand to be fixed!

Thanks!

vitormhenrique, 2017-05-03:
But did you removed the output and let only the temperature control? Can you enable debug information on the advanced area of the settings for the plugin and copy the log for me?

Also, do you have the active low / active high set correctly?

Atenciosamente,
Vitor Henrique