vitormhenrique/OctoPrint-Enclosure #36: Notifications of events

mike-berriman, 2017-04-28:
This is obviously an enhancement request.

I currently use a plugin that notifies me when prints start/stop etc via telegram messenger.  I'd love to get notifications (for example filament run out, smoke sensors etc) in the same way.

Can plugins in OctoPrint talk to each other?  Obviously, with the multitude of different plugins, it'd be hard to cater for them all.

Just a thought though.

Cheers,
Mike.

vitormhenrique, 2017-04-28:
Easiest way is actually for me to implement everything on my plugin.

I'll do some research and investigate adding this feature and put it on my roadmap.

vitormhenrique, 2017-05-12:
added on 3.5. I still need to create the documentation. I added using [IFTTT](http://ifttt.com/) this is much better because you can select how you want your notification, you can use the app, email, another app... twitter... 
Just need to setup a maker channel and configure the key on the plugin.

closed 2017-05-12 by vitormhenrique

vitormhenrique, 2017-05-24:
Steps to configure:

Login / Create an account on IFTT.

Go to My Applets, create a new applet and select Maker Webhooks after clicking "This". Give access to the maker channel. Select "Receive a web request" and you need to specify a name for the event, I used "printer_event" I think you can not have a empty space here.
For that "That" part I'm using a Send a notification from the IFTTT app (but anything else should work, like an email, another app push notification). For the content use the following: {{EventName}} : {{Value1}} .

After you set up your applet you can test it clicking on your username on the top right, services, click on the maker webhooks and click documentation.
A web page with your key and a test button will show up. Fill up the {event} with the same name that you gave it before, my case printer_event and on the "value1" write something like: This is a test.

Click test it, it should notify your client / send an email or whatever else you configured. My experience is that it took 5 minutes to start working, but after you get it working there it should work fine on the plugin as well.

If you got it working, just go back to the plugin enclosure, advanced section, enable the notification and add the IFTT maker key, and the event name that you gave. That is it.

dmachadol, 2018-03-13:
Change filament not work in Anycubic I3 Mega.
Reason: not implemented M0 M1 command and printer non stop.

Workarround "pause printing" option, and put some gcode in GCODE SCRIPT (buble plugin) like this:

[BEFORE PAUSE PRINT]

{% if pause_position.x is not none %}
M117 Paused!
G91    ; relative XYZ
G0 Z5 F1800 ; move Z slightly upwards
G90       ; absolute XYZ
{% if pause_position.f is not none %}G0 F7200{% endif %}
G0 X5 Y5 ; move to a safe rest position,
{% endif %}

[AFTER PAUSE PRINT]

{% if pause_position.x is not none %}
M117 Resuming Print..
G90 ; absolute XYZ
G92 E{{ pause_position.e }} ; reset E move back to pause position
; move back to pause position XYZ
{% if pause_position.f is not none %}G0 F7200{% endif %}
G0 X{{ pause_position.x }} Y{{ pause_position.y }} 
{% if pause_position.f is not none %}G0 F1800{% endif %}
G0 Z{{ pause_position.z }}
; reset to feed rate before pause if available
{% if pause_position.f is not none %}G0 F{{ pause_position.f }}{% endif %}
{% endif %}

Also there as a 2 scripts about change tool (al most in 1.2.6 version), can you add option to set.

Scripts are "Before tool change" and "After tool change" using this scrips aboid a pause user interaction and filament sensor enclosure interaction. 

An other option is create a 2 gcode scripts in enclosure plugin directly (a before action and after action), and user interaction confirmation between before an after execution code.

EddyMI3d, 2018-03-13:
I took the scrips from this site: http://docs.octoprint.org/en/master/features/gcode_scripts.html?highlight=pause
They work like charm with my Velleman K8200

vitormhenrique, 2018-03-13:
@dmachadol your complain is not valid for so many reasons:

1. **You commented on a topic that has nothing to do with filament detection**
2. The plugin ships with a generic g-code configuration that works for "marlin" firmware, you need to figure it out what to use on your installation and put that gcode on the advanced settings section of the plugin, the script needs to send a PAUSE command somewhere on it.