vitormhenrique/OctoPrint-Enclosure #37: GPIO Input

rbauwens1, 2017-05-02:
When adding an input for GPIO control, the event trigger field is set to "fall" even if the "rise" option is selected.

Would it also be possible to add functionality to shut off the heater/temperature control from the GPIO input?  I've added a smoke detector to my enclosure and would like to be able to shut off both the printer as well as the heater if smoke is detected.

(Also really minor but it looks like there's a typo in the plugin url in setup.py, so the octoprint plugin manager does not redirect to this github page properly.)

vitormhenrique, 2017-05-02:
what version of the plugin you have? 
This was a bug on the previous release, so I'm not sure if you have the latest....

vitormhenrique, 2017-05-02:
Also, what type of control are you adding? so I can test the same thing....

rbauwens1, 2017-05-02:
Ah sorry about that, the plugin version was out of date (version 3.43).  I've just updated to 3.45 and will test to see if the issue is resolved.

rbauwens1, 2017-05-02:
Thanks! I had to remove and re-add the input but it is working now.

closed 2017-05-02 by rbauwens1

vitormhenrique, 2017-05-02:
No problem, can you please create a new "feature request" with heater control using event triggers so I don't forget to add on future releases? 
Also, be sure to use external pull up / down resistors for critical things like this, as false triggers CAN happen using internal resistors....

rbauwens1, 2017-05-02:
Thanks for the tip, I'll keep that in mind for future input signals.  The smoke detector interconnect signal actually outputs 10V so I'm using a voltage divider to drop it to 3V for the pi, which should also act as an external pull down resistor.