vitormhenrique/OctoPrint-Enclosure #39: Adding Blob detection.

ChrisP-Git, 2017-05-02:
Hi,

I'm actually designing a blob detection ( https://github.com/ChrisP-Git/BlobDetector ) that detects if the print get unstuck from the bed and stick to the nozzle, which will create a plastic blob that can detroy extruder wiring. 

The thing is based on a laser barrier managed by an arduino that will trigger a digital port in case of obstruction. In this case, the print should be stopped. It's different from end of filament detection where the print should be paused.

What would be perfect would be an additionnal action with specific GPIO port.

Would that be possible, or even make sense to you ?

vitormhenrique, 2017-05-02:
I guess I just need to create another type of action with stop, instead of pause, and filament change.

Simple enough...

ChrisP-Git, 2017-05-03:
Great :) Take your time, there's no urgency, I do not have much free time to work on my detector right now, but I wanted to ensure that there's a viable roadmap
 ;)

vitormhenrique, 2017-05-12:
Added on 3.5

closed 2017-05-12 by vitormhenrique