vitormhenrique/OctoPrint-Enclosure #40: MQTT support?

mrwee, 2017-05-15:
Hi,
This plugin looks very nice and it seems like it extends OctoPrint with many features I could use. One challenge for me, is that I prefer to use MQTT driven external devices, rather than connecting via GPIO's on Raspberry. Is it possible to modify this to use MQTT commands instead?

vitormhenrique, 2017-05-23:
Definitely possible to add MQTT commands, for inputs and outputs. But I don't see a lot of gain, usually you need a raspberry pi connected to your printer in order to talk to your printer. And your controller is also around your printer, so it makes sense to use the gpio directly and avoid sending signals over the network to come back and do something 12 inches from your raspberry pi.
The only device that is more likely to be useful would be the power for the printer it-self with a smart outlet of some sort.
Also if I do implement what would you expect on the configuration? being able to set the topic and payload? or should I fix the payload with a simple API like "ON" and "OFF" for inputs and outputs. Hard coding the API would be easy to implement.
Also I would rater not implement reading the temperature via MQTT, in case of the client stops receiving data from the enclosure sensor it would not be able to control the temperature.
Another thing: EVENT driven MQTT events on the plugin don't make sense, because if you have MQTT devices on the network, they can already subscribe to those events and handle them accordingly. So I guess only outputs would be a real world useful enhancement.

floppystick6, 2017-06-13:
It would be nice to be able to control gpio via MQTT so my home automation hub can send a message to the octopi and it would turn the printer's main power on and turn off the printer and lights when the print ends. 

I would also be great if it would report back the status of the temperature and the heater's gpio status to track how often the heater is on and for how long (same goes for other gpio as well). And if it sent back the filament Sensor status, the the home automation hub could send a text message to my phone telling me it is out of filament. 

To implement it, I think it would be fine if you were to hard code the topics and payloads, both for the status and commands. I also think it would be best if it could use the octoprint MQTT plugin's "helpers", that way there is only one setup location for the MQTT server's info and credentials.

petonic, 2018-03-29:
I'd like to second Floppy's request for MQTT, and approach of using the MQTT plugin's "helpers" for config.  

Example, I have Tasmota installed on a Sonoff that controls mains to my 3d printer.  Instead of wiring the temp sensor to the RPI, I wired it to the Sonoff so that I could have a more direct path to shutting down mains power in case of runaway enclosure temps, and (granted: weird edge case) the RPI is down or non-responsive.

This looks like a very handy plugin, and I'm looking forward to using it soon.

xinuxnt, 2018-10-29:
I agree with petonic and floppystick6 with Mqtt add (you can implement a mosquitto on the same board) plus node-red you can open a big bunch of posibilities to an ultimate automation.

petonic, 2018-10-29:
Hi -- Related to this, I've implemented a self-contained Sonoff thermostat for an enclosure fan.  Basically, the enclosure fan relay Vcc is on one output pin of the Sonoff, and the DHT22 (or whatever) is on a sensor pin of the Sonoff.  You can plug in a range of temperatures for which you want the enclosure to remain within, i.e., 26 degC - 36 degC.  It implements a hysteresis mechanism.

This is done using Sonoff's relatively recent "Rules" mechanism.  When I'm back from my travels, I can post the code if anyone is interested...

ibishop100, 2018-12-07:
This would be a great feature to allow people running octoprint on a computer to use this plugin.  That's where I am at right now.  I have a PC to run two octoprints/webcams and would like to be able to use this plugin with my setup.

I was considering going through and changing all the direct hardware calls to serial calls to an arduino with the enclosure hardware hanging off.  Then I would just have to build up the Arduino code to run the enclosure.  If it were MQTT, I would use an ESP and half the work would be done.

Even more interesting would be if this plugin had an I/O plugin module allowing users to choose between RPI, MQTT, or serial to an arduino helper.  If that were to happen, I would be happy to build out the serial protocol and arduino code.

kaju666, 2019-02-17:
@vitormhenrique  hi. I have a question. Is there any "command" to power on/off added output's .
It will be nice that with plugin https://plugins.octoprint.org/plugins/mqtt/ i will be able to turn on my printer and lights using mqtt command.

vitormhenrique, 2019-02-18:
I don't know what do you mean by command, if the plugin can send gcode using mqtt it would be possible, other than that all the current API is documented on the wiki.

kaju666, 2019-02-20:
> I don't know what do you mean by command, if the plugin can send gcode using mqtt it would be possible, other than that all the current API is documented on the wiki.

Thanks, didn't see the API.

I made a script and instrucion for others interested:
https://github.com/kaju666/OctoPrint-Enclosure-Relay-MQTT-Integration

Exit2Studios, 2019-04-08:
Is there any progress in adding MQTT support for the input/output pins?  I would like to control my lights/fans/etc from Home Assistant.  Thank you!

pattonwebz, 2019-07-03:
I too would like to vote for addition of MQTT support, primarily for toggle on/off ability on the configured output pins.

I am willing to help with implementation, however I am limited with experience of python so some guidence will be required of anyone can give it.

pattonwebz, 2019-07-03:
As I am limited in experience with python I won't be messily be able to create an MQTT interface from scratch. I have found the `paho-mqtt` library which looks to be the most common choice for adding MQTT features in python applications.  @vitormhenrique wood to be opposed to

cnf, 2020-05-25:
the MQTT plugin mentioned earlier in this thread has hooks that could be used. I'l like to get sensor / state information pushed into MQTT myself, controlling the enclosure over MQTTis of less concern to me personally.

fspralja, 2020-12-23:
Hi, wanted to monitor the enclosure temperature for long prints during the night and get alerts through home assistant if some threshold is hit, so I quickly edited the SI7021.py file to get all the temperature sensor values to mqtt/HA, added to file end:
```
#install paho mqtt lib with: pip install paho-mqtt
import paho.mqtt.publish as publish
#if u need HA auto discovery uncomment following:
#publish.single("homeassistant/sensor/enclosure_temp_%s/config" % address,
#  payload=u'{"dev_cla": "temperature", "name": "enclosure_temp_%s", "unit_of_meas": "\N{DEGREE SIGN}C", "stat_t": "octoPrint1/temperature/enclosure_temp_%s"}' % (address, address),
#  hostname="192.168.X.X",
#  client_id="octoPrint1",
#  auth={'username':"usr", 'password':"pwd"},
#  port=1883)

publish.single("octoPrint1/temperature/enclosure_temp_%s" % address,
  payload=cTemp,
  hostname="192.168.X.X",
  client_id="octoPrint1",
  auth={'username':"usr", 'password':"pwd"},
  port=1883)
``` 

this is highly unoptimal but better to have some monitoring than none while the printer is in same space as us, probably need to get some new smoke sensors inside the enclosure as well... have few around but...

wernight, 2022-01-04:
Adding the sensor to the GPIO makes sense often however there are clear advantage of an MQTT solution:

- Xiaomi Mi temperature sensor and such can be placed anywhere very easily without any wire
- Home Assistant (or an ESP-32 flashed with ESPHome) can read those sensors and include their values is other automation (i.e. it can be used outside of OctoPrint)

OctoPrint could export the enclosure but that'd probably only work if OctoPrint natively supports enclosures as a core feature (or if the HA integration incorporated metrics exported by this plugin which seems a long stretch).

siparker, 2022-04-23:
Perhaps the addition of Sub plugins similar to the way PSU control has them might be the way to go. as not everyone will need MQTT then including the paho / other libraries by default would be unrequired. 
That said integration with the existing MQTT plugin https://plugins.octoprint.org/plugins/mqtt/ would be even better. 
Anyone with some basic octoprint plugin creation experience want to assist me with trying to put this together? send me a message. I have some Python 3 experience but 0 in relation to creating an octoprint plugin.