vitormhenrique/OctoPrint-Enclosure #42: I2C LCD

stanoba, 2017-06-06:
Hi, your plugin is awesome, but missing one thing. Is it possible to implement I2C 1604 or 2004 LCD and display temperatures and status of the enclosure? I'm thinking  about LCD next to hardware buttons in one case as enclosure controller.
Thanks

vitormhenrique, 2017-07-11:
I'll implement this on the next release, but expect it next month as I don't have much time this one...

stanoba, 2017-07-12:
That will be nice, thanks.

miguelmmm, 2018-08-09:
Hi, having support for I2C 1602 would be awesome!
I'm currently building a case to add the 2 relays (PSU and light) and 2 temperature sensors (enclosure and ambient). Being able to display the temperature sensor info in a display (in addition to the web interface) would be very useful when people are close to the printer.
I see it's been over a year since the last comment. I hope this idea is not dead.

vitormhenrique, 2018-08-09:
Nothing is dead, but I have very limited time to work on the plugin, this feature will require a lot of research and figure it out the best way to handle all the customization that this plugin allows.
I only work on this on my spare time (i have one full time job and another business to run). That being said, the plugin is open source, anyone else can fork, implement and commit any code, I don't need to be the only one implementing the new features.

nedla1969, 2019-11-05:
Did this ever get done?

vitormhenrique, 2019-11-05:
if it was, it would be a closed item, i guess you might have problems to read my previous comment as well.

nedla1969, 2019-11-05:
Things do get forgotten,, I was only asking,,,