vitormhenrique/OctoPrint-Enclosure #43: How to: Pause printer & executing gcode

Boby71, 2017-06-09:
I'm using a filamenr sensor and everything works well when I use "pause printer".
But there's a problem: The nozzle stays hot, the printer stops directly at the position where it got the signal. Means: Directly at the printed part.

My expectation for "pause printer" would have been:
- printer stops x/y motions
- print head will be raised (10-20mm would be enough)
- printer stops heating (because if I'm not at home, it shouldn't stay heated unattended)
- then I could manually change the filament (after manual preheating)
- and after loading the new filament I click on "continue" in Octoprint and the print head lowers down to the last print position - and continues printing

Currently this is not working for me this way:
- either it stops and stays heated (possibly for hours) or ["pause printer"]
- or it just ignores the filament sensor completely ["change filament"]

Is there anything I can do to achieve the behaviour I described?

Thank you!

sajovicd, 2017-06-10:
Did you write m600 for gcode execution after sensor is triggered.
You can also say to stop heating with gcode. Well what i think that is not neccessery, if your printer works for hours and hours without cooling down the nozzle, why there should be a problem waiting a couple more hours for you to get home.
Anyway you can remotely turnoff or do anything if you have pi connected to outer world.

vitormhenrique, 2017-06-10:
I guess you choose Printer pause for the filament sensor action, but you can select Filament Change and write a custom G-CODE on the advanced section of the plugin.

I use the following:

<pre><code>G91  ;Set Relative Mode 
G1 E-5.000000 F500 ;Retract 5mm
G1 Z15 F300         ;move Z up 15mm
G90            ;Set Absolute Mode
G1 X20 Y20 F9000      ;Move to hold position
G91            ;Set Relative Mode
G1 E-40 F500      ;Retract 40mm
M0            ;Idle Hold
G90            ;Set Absolute Mode
G1 F5000         ;Set speed limits
G28 X0 Y0         ;Home X Y
M82            ;Set extruder to Absolute Mode
G92 E0         ;Set Extruder to 0</code></pre>

vitormhenrique, 2017-06-14:
Did that fixed the issue for you?

vitormhenrique, 2017-06-19:
@Boby71 I'm closing this issue due to the lack of response. Please let me know if my post answered and fixed your quesrion.

closed 2017-06-19 by vitormhenrique

Boby71, 2017-06-23:
Sorry, I was on vacation and therefore far away from my loved printer.
But I found a simpler way since the new Marlin firmware added two commands for exactly this purpose: M601 & M602

My config now:
1) I leave "pause printer" as triggering event in enclosure_plugin
2) Follow this description: [https://shop.prusa3d.com/forum/original-prusa-i3-mk2-f23/anyone-tried-the-new-pause-print-gcode-m601-m602--t4604.html#p35660](url)
Thanks for you support @vitormhenrique