vitormhenrique/OctoPrint-Enclosure #44: NeoPixel Support

weltermarcel, 2017-06-11:
Hello, I've been trying to control a LED-Strip with your plugin and it doesn't seem to be working. I have communication with the Arduino but it doesn't seem to receive or the Raspberry to send anything. Does it work or is it still a work in progress?

I'd be happy to help with the Arudino code but I unfortunately know very little about coding for the RPi.

Thanks for the very useful plugin!
Marcel W.

vitormhenrique, 2017-06-12:
Are you using the firmware that I placed on the extras folder?

You also need to enable i2c on the raspberry pi. Follow this steps: https://learn.adafruit.com/adafruits-raspberry-pi-lesson-4-gpio-setup/configuring-i2c

After you configure, flash the correct firmware on arduino you can use the followig command to search i2c devices on the bus:

sudo i2cdetect -y 1

weltermarcel, 2017-06-12:
Yes, I'm using the provided firmware and have activated i2c communication. The address I receive is 4x04:

[image]

Is it normal that it is on the 4's column? If not, why do you think it could be doing it?

vitormhenrique, 2017-06-12:
Is correct!

Did you put 4 on the address field on the plugins configuration screen?

Also, did you corrected the pin that is being used on the Arduino on the neopixel?

Another thing is, enable debug information and send me the log...

Best regards,
Vitor Henrique

weltermarcel, 2017-06-13:
I've tried all possible combinations --> 0x04, 4x04, 04 but none of them seem to work. I will try it later again only with "4" because I'm not sure I have tried it like that. I still don't get how to read the table. What do the top row and the first column indicate? In the Arduino Sketch the address is 0x04 so shouldn't it respond on the first column?

Thanks for the help, I'll let you know later if it worked
Greetings,
Marcel

weltermarcel, 2017-06-13:
I've tried it with "4" and it still doesn't work... I've enabled the debugging on the arduino sketch but I don't receive anything, which makes me think that the raspberry isn't sending information. I'll try it later with another arduino board and if it still doesn't work I'll try to flash octoprint again and see if it works.

vitormhenrique, 2017-06-13:
Enable debugging on the plugin, and send octoprint log.

Best regards,
Vitor Henrique

weltermarcel, 2017-06-13:
Hi again,

I've reflashed octoprint and installed your plugin again and now it works... 

Thanks!

vitormhenrique, 2017-06-14:
Good... :)

closed 2017-06-14 by vitormhenrique