vitormhenrique/OctoPrint-Enclosure #45: GPIO pin numbering

marclefevre, 2017-06-21:
Thanks for the plugin.  It'll be just what I need.  For now, I'm just trying to control the PC power supply that supplies power to my printer.  I have configured GPIO 17 (board pin 11) as an output.  However, using the button, for that GPIO, in your interface does not change the output voltage.  So I must be doing something wrong.

To check my sanity, I went into Python and used the RPi.GPIO module to control the pin and that works.  If I configure GPIO.BCM, it's pin 17 that works.  If I configure GPIO.BOARD, it's pin 11 that works.

Can you figure out what I'm doing wrong?

I have version 3.51 of the plugin.

vitormhenrique, 2017-06-21:
First thing you need to do is enable debug on the advanced settings and send octoprint log here...

Does octoprint plugin interface report GPIO status? it should change the button to active once the gpio is toggled. Also if you have multiple plugins using GPIO that might be an issue changing the board type number, so you need to force it clear it first, and them change it and save.

vitormhenrique, 2017-06-21:
by default the plugin use BCM numbering when installed, so you should be using 17 on the plugin interface, no spaces.

marclefevre, 2017-06-21:
I have installed the filament sensor plugin but it's not configured yet.

I can't figure out where debug mode is set.  I don't see an advanced
settings menu selection anywhere.

vitormhenrique, 2017-06-21:
uninstall the filament sensor, this plugin handles filament as well.

go the the settings, and find the enclosure settings, go to the very bottom, and there will be and advanced link, click and it will show advanced settings.

marclefevre, 2017-06-21:
OK.  I thought that the debug control and advanced settings might be global
to octoprint so I was looking in the wrong place.  I found them now.

Debug was already enabled.
The "Use Board Pin #" option was checked.  So I unchecked it and pressed
the Clear GPIO mode button.
That still didn't fix the problem.  I don't see a way in octoprint to
observe the GPIO status.  Do you still want me to send some logs?

Thanks for all of the help.

Marc

vitormhenrique, 2017-06-21:
Best way is actually you send me an email, give your IP address, enable port forwarding 80 on your modem and create a temporary user name and password for me. So I can log on it and check it by myself.

And yes, send me the log.

marclefevre, 2017-06-21:
OK.  It's working now.  I changed the GPIO number from 17 to 11, cleared
GPIO and tried the interface buttons.  They didn't work (as expected).  I
then changed the GPIO pin back to 17 and cleared.  Now it's working.
Thanks for all of the help.

Marc

vitormhenrique, 2017-06-21:
No problem. Glad that you got it working! :+1:

closed 2017-06-21 by vitormhenrique

marclefevre, 2017-07-14:
Vitor,
I have an enhancement request for you.  I am going to try to run two
printers and enclosures off a single rpi.  To that end, I have connected 2
DS18B20 temp sensors and find that both work (looking in the /sys file
system).  However, I don't see a way to tell your plugin which of the two I
want to use with that instance of octoprint (i even opened the advanced
options section).  Can you add support for selecting which temp sensor you
want to your plugin sometime in the future, please?

Thanks,
Marc

marclefevre, 2017-07-15:
Hi Vitor,
I have one other enhancement request.  Since I'm using the DS18B20 for
temperature sensing, I don't have humidity sensing like I would with the
DHT11 or 22.  As a result, your temperature display shows 0% humidity.
Rather than display an incorrect humidity value, could you detect that
there is no humidity sensor and not display any humidity value?

Thanks,
Marc