vitormhenrique/OctoPrint-Enclosure #48: OctoPrint-telegram integration

leobel96, 2017-07-08:
Why don't you consider the integration with "OctoPrint-telegram" plugin integration adding the possibility to switch relay, see temperature,... from telegram bot? And what about adding the switch button on the navbar? Beautiful plugin, anyway!

vitormhenrique, 2017-07-11:
As far as I know, there is no way for two plugins to talk to each other, unless me and the telegram author get toguether and figure it out a way to implement this. I'll create an feature request with his plugin and see if it is possible or not.

leobel96, 2017-07-11:
Perfect, good idea!! I'll wait for news.

leobel96, 2017-07-11:
An idea should be to directly ask to @foosel

foosel, 2017-07-13:
A plugin like telegram could declare a [custom hook](http://docs.octoprint.org/en/master/plugins/hooks.html) to allow other plugins to provide additional commands or extend status reports and such. This is basically how the software update plugin (yes, it's actually a bundled plugin that takes care of updating stuff) also works, it [queries plugins providing a custom hook for update information](https://github.com/foosel/OctoPrint/blob/master/src/octoprint/plugins/softwareupdate/__init__.py#L96-L105).

vitormhenrique, 2017-07-13:
@foosel Thank you very much! that sounds good! But how would octoprint handle if the hook is not available? for example, someone install this plugin on an raspberry pi that does not have telegram plugin installed....

foosel, 2017-07-13:
The only thing that OctoPrint does with hooks is create a list which plugin *implements* what hook. But the hook's name is nothing OctoPrint cares about. So if something like the Telegram plugin were to define a custom plugin hook and some other plugin would declare it, stuff would match up. If said plugin was installed on an OctoPrint instance where the Telegram plugin is not installed, OctoPrint would still say "ah, that plugin provides a handler for this hook" but since nobody would be calling the hook (because only Telegram would and it's not there) nothing would happen. Simple as that.

There are some hooks that OctoPrint calls (e.g. the whole gcode processing stuff) and also some hooks provided by plugins bundled with OctoPrint (e.g. the software update config stuff), but if I removed one of those calls now, plugins having registered a handler for it would still just load fine. The only thing is that the hook would never be called anymore.

vitormhenrique, 2017-07-13:
@foosel got it! have I said that octoprint is awesome? thanks for this amazing work!

closed 2018-05-29 by leobel96

jacopotediosi, 2025-08-15:
FYI: this feature has just been implemented by https://github.com/jacopotediosi/OctoPrint-Telegram/commit/be139146d78b468dfa0a87435370766c56c045b8 and will be released in the upcoming 1.9.0 version.