vitormhenrique/OctoPrint-Enclosure #50: Start/Stop Time Support

CWRRT, 2017-07-11:
Is it possible to add start/stop time support based on actual time. For example, I have my lights turn on at start of print but would like them to go off at 10pm.

Thank you,
Cory Wiley

vitormhenrique, 2017-07-11:
Definitely possible, I'll only have time to implement next month or so....

vitormhenrique, 2018-03-01:
I just finished re-writing most of the plugin and **implemented this requested** feature, tested 95% of the plugin.

If you want to test it out uninstall the current plugin, go to the plugin manager, select get more, and where it says "from URL" use the development branch:

https://github.com/vitormhenrique/OctoPrint-Enclosure/archive/dev.zip

More documentation here:
https://github.com/vitormhenrique/OctoPrint-Enclosure/tree/dev

I'll push this to production after more people test it out.

CWRRT, 2018-03-02:
Just installed and will test out for a few days.

> On Feb 28, 2018, at 9:38 PM, Vitor de Miranda Henrique <notifications@github.com> wrote:
> 
> https://github.com/vitormhenrique/OctoPrint-Enclosure/archive/dev.zip <https://github.com/vitormhenrique/OctoPrint-Enclosure/archive/dev.zip>

CWRRT, 2018-03-02:
One thing I notice is that the auto turnoff only allows one or the other. In a perfect world, it would look like this.

Enclosure Lights
	Auto On during start of print
	Auto Off 60 seconds after print finishes OR turn off at 10pm, whichever comes first

vitormhenrique, 2018-03-02:
When you start a print, you actually have a pretty good idea when it will finish, so you probably would know what time of schedule you would want. I'm against creating a new check box, and a new field on the plugin, the configuration is pretty big as it is.
Maybe creating a separate schedule output that you could add to different outputs would be a better solution to accomplish this. 
But I'll not do it now, I spent a lot of time to get this release done, and I need to work on other projects...

vitormhenrique, 2018-03-07:
Implemented on V4.0

closed 2018-03-07 by vitormhenrique