vitormhenrique/OctoPrint-Enclosure #51: [Request] Navbar toggle button

leobel96, 2017-07-12:
Hi, It's me again, is it possible, in the next releases, to add toggle buttons for relay on the navbar like the "Octoprin PSU control" plugin? Maybe giving the possibility to choose which toggle add to the navbar and which not. Thank you!!

vitormhenrique, 2018-03-01:
I just finished re-writing most of the plugin and **implemented this requested** feature, tested 95% of the plugin.

If you want to test it out uninstall the current plugin, go to the plugin manager, select get more, and where it says "from URL" use the development branch:

https://github.com/vitormhenrique/OctoPrint-Enclosure/archive/dev.zip

More documentation here:
https://github.com/vitormhenrique/OctoPrint-Enclosure/tree/dev

I'll push this to production after more people test it out.

leobel96, 2018-03-01:
Thank you!!! Now I uninstall it I install the devel one.

leobel96, 2018-03-02:
@vitormhenrique I have installed it and it works fine, very good job!! Is it possible to add buttons on navbar as distinct buttons? I mean without having to click before on the cube

vitormhenrique, 2018-03-02:
Because now it is possible to add too much information on the navbar, I opted for the sub-menu system. 

It’s a compromise between going to the enclosure tab and not having the ability at all.

Best regards,
Vitor Henrique

leobel96, 2018-03-02:
Seems Logic. Time ago I opened anche issue about adding integration with telegram plugin. Are there news?

vitormhenrique, 2018-03-07:
Implemented on V4.0

closed 2018-03-07 by vitormhenrique