vitormhenrique/OctoPrint-Enclosure #52: Gap Control and over temperature protection

mac-za, 2017-07-16:
Thanks for the plug-in

I did something similar using Arduino, but am fairly new to the raspberry pi and Octoprint.

Is it possible to add two additional features?

- gap control - allow the temperature to drift by a user specified gap above and below the set temperature. This could potentially prevent short runs on the heating element

- allow the user to specify a maximum temperature and then either shutdown the heating element (and not use it again) or assume a worst case scenario and completely shutdown the printer

Thanks again

vitormhenrique, 2017-08-07:
Gap is not a good implementation for this scenario, the time of residence from the air heating the enclosure is so high that it takes a while to get to temperature, so in reality the heater will never turn on for less than a minute or so. 

The second request is possible to implement, for a work around today, you can have your temperature sensor triggering a IP pin on RPI and using that to shutdown the printer / power using the input GPIO's from this plugin.

vitormhenrique, 2018-03-01:
I just finished re-writing most of the plugin and **implemented this requested** feature, tested 95% of the plugin.

If you want to test it out uninstall the current plugin, go to the plugin manager, select get more, and where it says "from URL" use the development branch:

https://github.com/vitormhenrique/OctoPrint-Enclosure/archive/dev.zip

More documentation here:
https://github.com/vitormhenrique/OctoPrint-Enclosure/tree/dev

I'll push this to production after more people test it out.

vitormhenrique, 2018-03-07:
Implemented on V4.0

closed 2018-03-07 by vitormhenrique