vitormhenrique/OctoPrint-Enclosure #55: two extruders

alec220, 2017-07-30:
Hi!
I Try to use this plugin, but in my printers has two extruders. I did two sensors and connect its Pin 2 & 3.
1. How to explain plugin which sensor is swich on?
2. When in the first extruder finish filament sensor on and head go home heatbed too go down. I change filament, press Resume. Head go to right position and start print, but heatbed stay at home dont go up!
I use default script which was in your plugin.
Firmware Marlin for Ultimaker, Printer my, Alekmaker, (it looks as Ultimaker and use same parts.)

vitormhenrique, 2017-08-07:
Currently there is no way to disable sensors after you set them up. So basically you need to keep all sensors with filament to avoid them triggering, this could be improved on the future.

For the second question, you need to figure it out a g-code that works for your printer and your setup. The stock g-code works for most cases and printers, but clearly not for yours.

vitormhenrique, 2018-03-01:
I just finished re-writing most of the plugin and **implemented this requested** feature, tested 95% of the plugin.

If you want to test it out uninstall the current plugin, go to the plugin manager, select get more, and where it says "from URL" use the development branch:

https://github.com/vitormhenrique/OctoPrint-Enclosure/archive/dev.zip

More documentation here:
https://github.com/vitormhenrique/OctoPrint-Enclosure/tree/dev

I'll push this to production after more people test it out.

vitormhenrique, 2018-03-07:
Implemented on V4.0

closed 2018-03-07 by vitormhenrique