vitormhenrique/OctoPrint-Enclosure #59: Add more Temperature Sensors

4ntani, 2017-08-29:
Hi, 

thank you so much for the nice job you do with this plug-in that save my life :D 
What do you think about adding one or more Temperature Sensors for control the temperature of the stepper motors?

I think that is a good idea... 

If you think that i should have no paranoias with its temperature i will feel better :D

Thank you so much for the reply...

Andrea.

vitormhenrique, 2017-09-01:
Adding new temperature sensors might be possible, but depend on which one are you going to use. DHTs already cause problems when reading directly from the gpio.
One wire temperature sensors, and i2c would be best, most i2c sensors would be able to get four different address, so up to 4 temperature sensors.
The only problem is that realistically those temperatures would be only "for fun" not really usable, there is nothing you could do to react to a stepper motor heating up, only shutting down the print. But do you really would like that? Stepper motors do heat a lot, but are not a fire hazard. If they overheat they will basically fail and break. 
Current plugin is possible to add a small and cheap arduino with temperature sensors monitoring all stepper motors, and use one output to the plugin. If any stepper motor overheat, you write to the output and get the plugin to react, stopping the print.

4ntani, 2017-09-01:
Thank you so much for the fast reply :D 

I had not thought about arduino :D Nice idea!!! If i will have overheat problems i'll use it :D

FGMarshall, 2017-09-18:
If the room is too cold I use this to turn on the heater (Thanks!)
But in summer it can get too hot (my system is in a shed for fire safety)
I can use the alarm to turn a cool air fan on - but how to turn it off if it gets too cool?
So I need a second thermostat.
Please. . . 
Or where in the code can I tinker with this

vitormhenrique, 2017-10-08:
@FGMarshall I don't quite understand your use case to get two temperature sensors.
You can toggle between heater vs cooling on the supper / winter. 
Use different GPIO for a fan and for the heater.

Axel-Germany, 2017-10-08:
Thank you for that awesome Plugin. That is the best Plugin. I would like to use a MAX31855 breakout board from adafruit to controll a K-type termocoulper to check the hotend. That would be awesome and i would have a safer printer with that. Would be great if you could try to add that to the plugin.

vitormhenrique, 2017-10-09:
@Axel-Germany could you create a new feature request to add MAX31855 compatibility?

Axel-Germany, 2017-10-09:
Yes I would like to use the MAX31855 breakout board from adafruit https://www.adafruit.com/product/269and use the K-Type thermocouple to control the hotend temperature. If something goes wrong with the temperature i can use the plugin to switxh offthe printer so i add some safety to the printer when i am not there or the gcode just have a wrong temperature and it could get dangerous. The k-type sensor can take readings up to 1350 C so it will be more reliable then the sensor from china in my printer

pyip, 2018-01-12:
Any merit in sticking a DS18B20 onto the heatsink (radiator block) to measure potential heat creep? I'm thinking that may help in troubleshooting clogs due to heat creep.

vitormhenrique, 2018-01-12:
DS18B20  temperature range is only -55 to 125°, and I don't see how you could easily measure heat creep, only the radiated heat from the extruder to the sensor would mask the actual temperature of the heat sensor. You would need thermocouples for that.

vitormhenrique, 2018-03-01:
I just finished re-writing most of the plugin and **implemented this requested** feature, tested 95% of the plugin.

If you want to test it out uninstall the current plugin, go to the plugin manager, select get more, and where it says "from URL" use the development branch:

https://github.com/vitormhenrique/OctoPrint-Enclosure/archive/dev.zip

More documentation here:
https://github.com/vitormhenrique/OctoPrint-Enclosure/tree/dev

I'll push this to production after more people test it out.

bumbalu, 2018-03-04:
As you said above I use additional arduino board and temp sensors on each motor in enclosure. If motors get overheated the Peltier modules attached to the motors are turned on as well as fans on them. Works nice.

vitormhenrique, 2018-03-07:
Implemented on V4.0

closed 2018-03-07 by vitormhenrique