vitormhenrique/OctoPrint-Enclosure #63: Possibility to send custom gcode to printer

SirYwell, 2017-09-24:
I would like to be able to send custom gcode to my printer by pressing a button which is connected via GPIO. Examples: Set a temperature or moving the Z axis of the printer. Could you implement something like that?

vitormhenrique, 2017-10-08:
That is a good idea, and it was already on the pipeline, it will be on the next release.

vitormhenrique, 2018-03-01:
I just finished re-writing most of the plugin and **implemented this requested** feature, tested 95% of the plugin.

If you want to test it out uninstall the current plugin, go to the plugin manager, select get more, and where it says "from URL" use the development branch:

https://github.com/vitormhenrique/OctoPrint-Enclosure/archive/dev.zip

More documentation here:
https://github.com/vitormhenrique/OctoPrint-Enclosure/tree/dev

I'll push this to production after more people test it out.

vitormhenrique, 2018-03-07:
Implemented on V4.0

closed 2018-03-07 by vitormhenrique