vitormhenrique/OctoPrint-Enclosure #64: Feature Request: MQ9 CO Gassensor for smoke detection

Quas7, 2017-09-26:
Hi,

I like this plugin very much. A DHT11 was installed within 2 minutes.

It would be great, if the threshold adjustable digital output of an MQ9 sensor  could be used in the plugin to detect smoke. It is a plain 5V active low signal (high = CO below trigger, low = CO above trigger) - of course, a 3V3 voltage divider is needed for the Pi.

A little [Datasheet](http://www.mouser.com/catalog/specsheets/Seeed_101020045.pdf)
Cheap MQ9 sensor [board](https://www.amazon.de/gp/product/B06XSMCVJ9/ref=oh_aui_detailpage_o04_s00?ie=UTF8&psc=1)

These sensors also have an analog output but that would require an i2c-ADC for the Pi or some GPIO time-constant hacking to get a poor-mans-ADC. 
[Cheap ADC for the Pi](https://www.amazon.de/gp/product/B01ER6FK6G/ref=oh_aui_detailpage_o02_s00?ie=UTF8&psc=1)

I currently use an Arduino Pro Micro to read the analog MQ9 value and push it to the Pi via an usb-serial interface (yes, overkill). If the rate of change exeeds a certain value or a upper threshold is hit, the power supply is cut.

vitormhenrique, 2017-10-08:
Hello, sorry for the delay on the response. Well, today the plugin can already respond to digital inputs. 

As you already mentioned analog it is not possible without an ADC. but if you use an opamp or a comparator, or even the digital output of the sensor, you can connect it to any gpio, and on the setting put it as input, you can controls events for this input, it can send you messages using ifttt, connect other gpio anything that you want, it is already there ;)

Quas7, 2017-10-10:
Oh, missed that! I will test this. Thanks 👍

closed 2017-11-17 by vitormhenrique