vitormhenrique/OctoPrint-Enclosure #65: Toggle for Autostart

Tsalokin, 2017-09-30:
My printer is in my room and so I like it to shutoff when it's finishes at night, but I also sometimes print continuously which becomes a problem if I forget to turn off auto shutoff.
Toggling auto shut off is a little out of the way, it would be nice to have a toggle next to the buttons in the Enclosure Plugin tab. Maybe even an option to add or remove the button in settings, if it's not necessary for others.

vitormhenrique, 2017-10-08:
but do you want a button on the plugin page that turns the printer off? or a button that changes the setting?
I need to do some research, i don't think it is possible to change the setting from other pages, it might, but I don't know how...

vitormhenrique, 2018-03-01:
I just finished re-writing most of the plugin and **implemented this requested** feature, tested 95% of the plugin.

If you want to test it out uninstall the current plugin, go to the plugin manager, select get more, and where it says "from URL" use the development branch:

https://github.com/vitormhenrique/OctoPrint-Enclosure/archive/dev.zip

More documentation here:
https://github.com/vitormhenrique/OctoPrint-Enclosure/tree/dev

I'll push this to production after more people test it out.

vitormhenrique, 2018-03-07:
Implemented on V4.0

closed 2018-03-07 by vitormhenrique