vitormhenrique/OctoPrint-Enclosure #66: Feature Request: Variable PWM Output based on Temperature

mantisrobot, 2017-10-04:
Hi,

This is a great plugin, but I wonder if its possible to have a variable PWM output based on a temperature between to set points. This is to control a PWM based fan to cool the cabinet when it is too warm, but something more sophisticated than bang bang control. Possibly with:

Minimum Duty
Maximum Duty
Start Tmperature
End Temoerature

Would also be good to have some way to set the fan on 100% at the end of a print to cool the enclosure and extract fumes.

Cheers

Matt.

vitormhenrique, 2017-10-08:
I had few people request the same thing, I can implement this, but honestly think it will not be as good as you might think for a few reasons.
1 - The on/off control is good enough for the cooling feature because the time of residence for heating the enclosure it big, after it gets to the temperature, it will remain there for a few minutes, so if you want to cool it you will want as much fan as you can.
2 - pwm controlling a fan will require you to use a FET to connect the fan, a relay will not work of course. 

Do you have any data, or a specific setup that you can tell me, that would make PWM control based on the temperature more sense?

mantisrobot, 2017-10-08:
Hi,

You may be correct, the reason I think a proportional speed fan would be good Is

A) to stop sudden cold air drafts being drawn in by the fan in bang bang control mode.

B) to keep some air flow int the cabinet even if it's slight, so as to remove fumes. I would connect the output of my fan to an extractor tube and vent to outside.

This may be achievable using bang bang type control, however use a PWM output that allows the user to set a minimum PWM duty, and an on state PWM duty at the set temp.

I was going to use 4 wire PC cooling fans that have the drive circuit built in. You supply them with 12v and a pwm duty signal at about 20khz in order to control the fan speed.

The function at the end of the print to switch the fan fully on would also help cooling the part quickly and remove fumes.

mantisrobot, 2017-10-13:
Hi,

I have managed to get the functionality I needed the following way:

I use a Corsair 120mm 4 pin PC cooling fan. The fan has a PWM speed control input. The input usually has a 20Khz PWM signal to control speed, however, simply setting this pin to logic level high or low will give you the fan running at full speed or minimum speed which seems to be about 10%. Using this feature I have done the following:

Set temperature control mode as a "heater" with trip point of 35 degrees, with an inverted output, e.g. low level "on". Connect this output to the PWM input of the fan, and a separate 12v DC supply. The tacho pin is unconnected.

Now when a print starts assuming cabinet is below 35, the "heater" output is on, logic low, so the fan is on but at its lowest setting which is enough to extract fumes without pulling too much heat from the cabinet. Now as the temperature rises and goes beyond the trip point, the "heater" output goes high which puts the fan full on and starts cooling the cabinet. I have got the cabinet within +/- 1 degree using this method. Now crucially when the print is finished the "heater" output goes off, logic high which leaves the fan on full to extract fumes and heat cooling the cabinet and print.

vitormhenrique, 2018-03-01:
I just finished re-writing most of the plugin and i**mplemented this requested feature** with a bunch of extra features, tested 95% of the plugin.

If you want to test it out uninstall the current plugin, go to the plugin manager, select get more, and where it says "from URL" use the development branch:

https://github.com/vitormhenrique/OctoPrint-Enclosure/archive/dev.zip

More documentation here:
https://github.com/vitormhenrique/OctoPrint-Enclosure/tree/dev

I'll push this to production after more people test it out.

stuemi, 2018-12-10:
Hello,

I need the same functionality like mantisrobot. I would like to use a 4Pin fan too. 
Can I connect the fan as shown in the attached picture? I would like to turn on/off the fan with one GPIO Port in connection with a relay. To control the speed of the fan I want to connect the PWM Cable to an another GPIO Port. Is this possible? Thanks for your answer.

[image]