vitormhenrique/OctoPrint-Enclosure #67: orange pi zero - error with wiringOP-zero! - GPIO.setmode

coolduke666, 2017-10-08:
the orange pi zero is an optimal octoprint server - includes wifi and is very fast, small and cheap and there is also an "wiring pi" equivalent for the orange pi, namely https://github.com/xpertsavenue/WiringOP-Zero

sadly your plugin does NOT work on the orange pi zero - there is a problem with the GPIO:

2017-10-08 07:56:46,675 - octoprint.plugins.enclosure - WARNING - An exception of type RuntimeError occurred on updateOutputUI. Arguments:
('Please set pin numbering mode using GPIO.setmode(GPIO.BOARD) or GPIO.setmode(GPIO.BCM)',)

I don't know the effort it requires, but it would be great if you support orange pi as well - your plugin is really great!

As an alternative your can simply add text-field to invoke shell-scripts once the gpio-has to be switched... because the it will work perfectly...

if you need someone for testing, please let me know!

thx!

vitormhenrique, 2017-10-08:
The documentation of this plugin is clear: it only is supposed to work on Raspberry Pi family. 
That been said, I had a lot of requests to make it work on orange pi, vide @marcoamr2002 issue, and other emails that I got. 
I'm willing to make the changes and test the implementation of the Orange Pi. But I do not have an orange pi board, and I'm not going to buy one just for this purpose. So either someone donate / lend me one board so I can proper test and implement, or we will need to wait until someone does the changes and push it to my github. 
On the past, few people asked me to implement other temperature sensors and they shipped me the sensors so I could test them. It is almost impossible to make the changes and rely the community to test, I tried this before with no luck.

coolduke666, 2017-10-08:
thanks for the feedback. this is the answer I expected ;)

Is there anybody out there who can adapt the plugin to orange pi? would be highly appreciated!

mcmasterp, 2017-10-08:
@coolduke666 why not help get this going and support the developer at the same time by sending him an orange pi or a gift card for one? If you want to be a super awesome supporter send him enough to get one and something else to solve another users request or just extra for him to spend on something completely unrelated.

coolduke666, 2017-10-08:
as a general approach you could simply add options for shell scripts... so the plugin will be not only for raspberry.

eg: textfields for scripts: 
gpio on
gpio off
gpio state - return code as state

when turning gpio on/off your plugin has to invoke the script on or off. in the simple shell script everybody can utilise the proper gio implementation which works every time.

could you please implement that simple approch. so either direct switching via python GPIO or via shell script.

good idea?
thx

coolduke666, 2017-10-08:
@mcmasterp simple because I only have one and shipping another from china will take 6+ weeks...

mcmasterp, 2017-10-08:
@coolduke666 no problem Amazon to the rescue: Orange Pi One Single Board Computer Quad Core 1.3GHz 512MB DDR3 with Case and Heatsink https://www.amazon.com/dp/B01CXE2SMS/ref=cm_sw_r_cp_apa_4EL2zb2CA63PM

Good price and only two days shipping! 

Best part is you can send an Amazon gift card through email super easy and a little extra could pay for something really cool he could use.

vitormhenrique, 2017-10-09:
@coolduke666 shell scrips won't be a viable solution, because of two things:
1) This plugin uses interrupts for most of it's features, you can use as filament sensor, button to react and so something, turn on and off other gpios, smoke sensor, etc... If I use shell scripts all that functionality will be lost, and I would have to maintain two different "ways" to handle only the heater and cooling, it is just not worth. 
The first release was using shell scripts and I changed to pure python.
2) Another issue is with security, octoprint does not run as sudo, so shell scripts inside the python don't have enough security to access the gpio pins, you need to manually add octoprint to gpio access and tweak a bunch of security issues.

coolduke666, 2017-10-09:
I wrote also a comment to the owner of the wiringpi zero git (https://github.com/xpertsavenue/WiringOP-Zero)... maybe he can fix this issue easily, because it look like there is a slight incompatibility with the original wiring pi for raspberry.

coolduke666, 2017-10-09:
@mcmasterp a good idea in general... firstly I am located in europe, therefore I can't use amazon.com, I only can purchase from amazon.de, who only delivers to germany and austria.

secondly, I am a poor student and have to look at my spendings. that is the reason why I purchased a far cheaper orange pi zero, instead of a raspberry pi 3 (10 euros vs 35 euros).

when looking at your website, I realised that you are a successful business man and maybe you could help out and send the amazon gift card to the the owner - that would be great and highly appreciated by a lot the orange pi users.

mcmasterp, 2017-10-09:
@coolduke666 those are all very good reasons. I hope your studies go well and I like that you tinker with 3d printers and pi's at the same time, we need more people like you.

I did consider sending the orange pi myself since your not the first to ask about it. At the moment I have zero dollars as is the life of a self employed person but if my luck changes as it does from time to time it's on my list now. Let's both hope for a upswing soon lol. Make that all three of us, you me and my wife lol

vitormhenrique, 2017-10-09:
Guys, lets keep the conversation on this page related to the plugin.
I had dedicated a lot of my free time on this plugin. And this specific change, despite I don't think would require a lot of coding it will require a lot of testing, more than 8 hours to setup and every single possibility and proper test the plugin. I have a full time job, and I do all this on my free time to help everyone, I don't usually ask for donations. If I had 25 dollars spare, I would get an orange pi and implement this, but now I can't.
If 4 people contribute 6 bucks they could get an board and ship it to me. If you can't that is fine as well, you will have to wait until I can get the board / or someone ship one to me.
If you have a board, you can also learn to code, learn the octoprint development documentation, make the changes and push to my github. 
There are a lot of ways to help.

mcmasterp, 2017-10-09:
My apologies. Will do

xinuxnt, 2018-10-29:
If you put a "Donate" button somewhere I guess you Will got the money for that. I have my dollar bill ready :-)

vitormhenrique, 2018-11-08:
@xinuxnt there is an "buy me a coffee" on the main page ;)

xinuxnt, 2018-11-08:
Oh yes, but that is for you and your effort not to buy things for others!
👍    You know, I mean something apart from who was interested on an Orange
Pi implementation.

Kiron21, 2019-05-20:
Hi to everybody.

Any progress? :)

Kiron21, 2019-05-21:
Hi again.

It seems to work!!

1) install OPi.GPIO  https://pypi.org/project/OPi.GPIO/  under your octoprint python version

`sudo runuser -l  octoprint -c '/home/octoprint/OctoPrint/venv/bin/pip install --upgrade OPi.GPIO'`

2) make group gpio and insert your octoprint user into:

`sudo groupadd gpio
sudo usermod -aG gpio octoprint`

3) enable using GPIO in userspace:

`sudo nano /etc/udev/rules.d/99-gpio.rules`

and insert:

```
SUBSYSTEM=="gpio", ACTION=="add", RUN="/bin/sh -c '\
  chown -R root:gpio /sys/class/gpio && chmod -R 770 /sys/class/gpio;\
  chown -R root:gpio /sys$DEVPATH && chmod -R 770 /sys$DEVPATH"`
```

3) Change file of enclosure plugin:

`sudo nano /home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_enclosure/__init__.pyb
`

and replace line: `import RPi.GPIO as GPIO`
to:  `import OPi.GPIO as GPIO`

5) Restart and test. You have to found goog pin numbers. For Example: I am using physical pin 8, in BCM it means 198 (from OPi.GPIO library source code), GPIO is numbered as 28 and in Python is in fact **14** :)

IInickII, 2019-09-18:
Hi to all,

I was trying steps above but my problem is that my file of enclosure plugin is empty. I was also trying to reinstall the plugin but with no luck 0 still the same. Can you help me please? I have Orange Pi Lite.

Thank you

xinuxnt, 2019-10-06:
Hello  IInickII

With sudo nano /home/octoprint/OctoPrint/venv/local/lib/python2.7/site-
packages/octoprint_enclosure/__init__.py

Please comment if You have success.

IInickII, 2019-10-07:
Hi, I solved it before but thank you for your reply. I also edited pin mappings for me let it match pin numbering of the board.

kokospalme, 2024-03-22:
Hello, did someone get it work on an Orange pi zero 2 in 2024?
When I install the plugin, nothing seems to happen... It isn't even listed in the installed plugins.
I don't have much money but I think I can give about 10, maybe 15€ if that would help you developing this great plugin for the orange pi zero 2 :) 
My OS: ```Orange Pi 3.0.6 Bullseye with Linux 5.16.17-sun50iw9```