vitormhenrique/OctoPrint-Enclosure #68: After update to 1.3.5 target temp shows dash

Flappie, 2017-10-17:
After update to 1.3.5 the target temperature in the enclosure temp settings fields shows a dash.
It does not show the target value.
Value does register and plugin is working as expected switching output pin when set target temp is hit.

[image]

Searched the issue list could not find this issue.
Please correct if this is a duplicate.

vitormhenrique, 2017-10-18:
definitely not duplicated issue, I'm very very busy this week preparing for Houston maker faire and did not upgrade my octoprint yet. 

I'll fix this issue and start working on new features next week.

Flappie, 2017-10-19:
No worries, as said the switching on the target temperature does work fine.

vitormhenrique, 2018-01-22:
I fixed this on the dev branch, I'm also working on this fixes and  enhancements, if anyone can change to the dev branch and help testing let me know...

vitormhenrique, 2018-03-01:
I just finished re-writing most of the plugin and **implemented this requested** feature, tested 95% of the plugin.

If you want to test it out uninstall the current plugin, go to the plugin manager, select get more, and where it says "from URL" use the development branch:

https://github.com/vitormhenrique/OctoPrint-Enclosure/archive/dev.zip

More documentation here:
https://github.com/vitormhenrique/OctoPrint-Enclosure/tree/dev

I'll push this to production after more people test it out.

vitormhenrique, 2018-03-07:
Fixed on 4.0

closed 2018-03-07 by vitormhenrique