vitormhenrique/OctoPrint-Enclosure #69: Plugin Does Not Support `X-Script-Name` Option For Reverse Proxy

randuhmm, 2017-10-19:
I am using a Nginx server to reverse proxy to my OctoPrint server and I was able to follow the guide here to properly setup a different `baseUrl`:

https://github.com/foosel/OctoPrint/wiki/Configuration#server

It seems that this plugin does not honor this configuration and still uses the root path for http requests. For example:

`/plugin/enclosure/...`

vitormhenrique, 2017-10-19:
I never tried to use the plugin using those configurations, I have to do some research to fix this issue, If you could I would suggest using the default configuration for now to get it working...
I'll take a look on the next release and see if I can fix it.

randuhmm, 2017-10-20:
I dug around a little bit and i believe that I have located the root cause of the issue. It seems that the flask routing properly updates the base url for plugins so this is simply a client-side issue. If the `X-Script-Name` header is passed with a value of `/my-base`, then the flask webserver properly sets the base path for all plugins such that the path will look like this:

`/my-base/plugins/<plugin>`

It seems that there is a globally accessibly variable that can be used to access this location in JavaScript so that any XHR calls can be properly constructed, the variable is:

`window.PLUGIN_BASE_URL`

And, in the above case it should have the value:

`/my-base/plugin`

(I may have gotten a few of the details wrong, I am riding the train right now and working off memory haha)

I don't mind taking a stab at this issue if you like. Thanks for your work on this plugin, I find it quite useful!

closed 2017-12-18 by vitormhenrique

randuhmm, 2017-12-18:
👍