vitormhenrique/OctoPrint-Enclosure #73: Change FIlament Script not working.

cpmorris, 2017-11-17:
Im not sure why but when the script is pushed its pushing as lowercase(even though its typed in caps in the plugin setup).  Terminal wont recognize some commands in lower case.

cpmorris, 2017-11-17:
[image]
[image]

cpmorris, 2017-11-17:
Specifically the x y coordinates.  If I type them in caps directly in the terminal it works fine.  Thanks

vitormhenrique, 2017-11-17:
you are right.

I just noticed that there is an issue on the code...

https://github.com/vitormhenrique/OctoPrint-Enclosure/blob/master/octoprint_enclosure/__init__.py#L436

I'll fix on the next release.

vitormhenrique, 2017-11-17:
if you want to quick fix, just ssh to your rpi, find the line that I'm pointing and change it to upper instead of capitalize...

cpmorris, 2017-11-17:
Any idea on timing?  I have some 60hr prints that exceed my spool quantity so this will be pretty handy. THanks

cpmorris, 2017-11-17:
oh ok ill see if I can figure that out.

cpmorris, 2017-11-17:
What and where is the file name to get the script?  I'm not familiar enough with the file structure on these yet but can get it swapped wonce I know where its at.

vitormhenrique, 2017-11-17:
if you have a standard octopi installation, using the image pre-created for raspberry pi the file should be here:

/home/pi/oprint/lib/python2.7/site-packages/octoprint_enclosure/

just edit the file __init__.py with the changes that I said, and restart your system.

cpmorris, 2017-11-17:
One last question...what's the command for opening it in an editor?  Linux command line is foreign to me but I have located the file.

vitormhenrique, 2017-11-17:
you need to open the file using a text editor, if you have zero experience editing files in linux search how to use nano.
it is not quite opening, using the mouse and changing the file.

cpmorris, 2017-11-17:
yeah i had found nano but was having trouble oprining file....the file name has two underscores before and after

cpmorris, 2017-11-17:
good to go thanks for the help!

vitormhenrique, 2017-12-18:
fixed on https://github.com/vitormhenrique/OctoPrint-Enclosure/commit/bf6efc2cd1b7f06e09aa5b429f5fa0851ba19b77

closed 2017-12-18 by vitormhenrique