vitormhenrique/OctoPrint-Enclosure #74: Hum: in navbar with DS18B20 at 0.0%

j4c3, 2017-11-23:
The DS18B20 doesn't support humidity, right?  It would be nice if we could disable humidity in the navbar for devices that don't support it.  Or, a toggle to suppress it.

Also, a sidenote, the DS18B20 can be used on pins other than 4.  In my boot/config.txt I use the following to enable OneWire: dtoverlay=w1-gpio,gpiopin=17.  The enclosure plugin is able to use it without issue. 

Thanks!

LeandroBarros, 2017-12-13:
+1 for adding this. Would be a nice addition. Thanks

vitormhenrique, 2017-12-18:
I'll implement on next release

vitormhenrique, 2018-03-01:
I just finished re-writing most of the plugin and **implemented this requested** feature, tested 95% of the plugin.

If you want to test it out uninstall the current plugin, go to the plugin manager, select get more, and where it says "from URL" use the development branch:

https://github.com/vitormhenrique/OctoPrint-Enclosure/archive/dev.zip

More documentation here:
https://github.com/vitormhenrique/OctoPrint-Enclosure/tree/dev

I'll push this to production after more people test it out.