vitormhenrique/OctoPrint-Enclosure #75: Direct NeoPixel Support

dismaster, 2017-11-23:
As NeoPixel is directly supported via the PI would it be possible to add a choice if it's controlled through raspberry or arduino?

Please see attached python script with which you can control it directly - it required the Arduino library from here: http://frederickvandenbosch.be/?p=1014

thanks!
[neopixel.py.txt](https://github.com/vitormhenrique/OctoPrint-Enclosure/files/1500288/neopixel.py.txt)

vitormhenrique, 2017-12-18:
I initially coded direct support, but, neopixel and raspberry pi, despite "work" is unreliable. 

I'll add as an option on next revision, but I discourage using it.

dismaster, 2017-12-18:
Great stuff! Would be really great to get it!

vitormhenrique, 2018-03-01:
I just finished re-writing most of the plugin and **implemented this requested** feature, tested 95% of the plugin.

If you want to test it out uninstall the current plugin, go to the plugin manager, select get more, and where it says "from URL" use the development branch:

https://github.com/vitormhenrique/OctoPrint-Enclosure/archive/dev.zip

More documentation here:
https://github.com/vitormhenrique/OctoPrint-Enclosure/tree/dev

I'll push this to production after more people test it out.

vitormhenrique, 2018-03-01:
@comradequestion, you don't need to use your fork now, you can try installing the dev branch and check the other new features as well.

vitormhenrique, 2018-03-07:
Implemented on V4.0

RitchRock, 2018-04-29:
Is there anything else that needs to be installed for this to work?

Perhaps it's not supported on RasPi 3 B+.

vitormhenrique, 2018-04-29:
Follow this guide:
https://learn.adafruit.com/neopixels-on-raspberry-pi?view=all

If it works

closed 2018-04-29 by vitormhenrique