vitormhenrique/OctoPrint-Enclosure #76: Only reads GPIO when user is sudoer

ThierryDi, 2017-12-05:
Hello,
Issue is the following: by default the GPIO is accessible only by sudoers, hence if the user running octopi is not a sudoer with "no password" (which is not very security friendly) the GPIO read doesn't work.
Regards,
Thierry

vitormhenrique, 2017-12-05:
This is not an issue with the plugin, and there is nothing I can do about it. 
The plugin uses RPi.GPIO library to access the GPIO, you can just fix the permissions to the memory and gpio folders to the user that is running the server. The standard octopi image works fine.

Check [this](https://raspberrypi.stackexchange.com/questions/40105/access-gpio-pins-without-root-no-access-to-dev-mem-try-running-as-root) for more information:

closed 2017-12-05 by vitormhenrique

ThierryDi, 2017-12-06:
Hello,
After fixing the gpio access the test using the Adafruit example works with no sudo, thanks for the tip.
BUT the plugin still doesn't read the temp (says 0).
Log: 2017-12-06 18:32:04,196 - octoprint.plugins.enclosure - WARNING - An exception of type ValueError occurred on readDhtTemp. Arguments:
('need more than 1 value to unpack',)
I run a self installed octoprint instance, therefore the user is not pi, but this shouldn't matter,right?
Thanks in advance

vitormhenrique, 2017-12-06:
It is not getting the temperature from the plugin, probably because it can not find the scripts, or the access for the files are not correct:

Check if the files were copied, the default installation would be something like this:
/home/pi/oprint/lib/python2.7/site-packages/octoprint_enclosure/getDHTTemp.py

It should not need to run as pi, but again, if you just flash the default image it's going to just work, I suggest doing that. If you don't want to to this, find the installation files, check that they are there, fix the permissions, test the scripts by using:

sudo python /home/pi/oprint/lib/python2.7/site-packages/octoprint_enclosure/getDHTTemp.py 22 4

Also, you can just add a simple LED to raspberry pi GPIO and test a simple output, to see if it's working.

ThierryDi, 2017-12-06:
Thank you for your answer the files are here, they are in /home/MYUSER/OctoPrint/venv/lib/python2.7/site-packages/octoprint_enclosure
and the script works fine, even without sudo (I just run python getDHTTemp.py 22 4 , it returns temp / hum)

I guess the call to the script expects the path to be /home/PI/OctoPrint/venv/lib/python2.7/site-packages/octoprint_enclosure,right?

vitormhenrique, 2017-12-06:
Actually it does not, it should automatically find the file where the plugin is installed:

https://github.com/vitormhenrique/OctoPrint-Enclosure/blob/master/octoprint_enclosure/__init__.py#L228

Enable the debug mode on the advanced part of the plugin, it will log where it thinks the files are and more info to trouble shoot. Reboot raspberry pi, send me the entire log after you start getting the errors again.

ThierryDi, 2017-12-06:
There’s the log with debug mode enabled:

2017-12-06 19:48:18,992 - octoprint.server - INFO - ******************************************************************************
2017-12-06 19:48:18,993 - octoprint.server - INFO - Starting OctoPrint 1.3.5 (master branch)
2017-12-06 19:48:18,994 - octoprint.server - INFO - ******************************************************************************
2017-12-06 19:48:19,168 - octoprint.plugin.core - INFO - Loading plugins from /home/titi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.5-py2.7.egg/octoprint/plugins, /home/titi/.octoprint/plugins and installed plugin packages...
2017-12-06 19:48:22,673 - octoprint.plugins.discovery - INFO - pybonjour is not installed, Zeroconf Discovery won't be available
2017-12-06 19:48:22,687 - octoprint.plugin.core - INFO - Found 8 plugin(s) providing 8 mixin implementations, 5 hook handlers
2017-12-06 19:48:22,757 - octoprint.server - INFO - Intermediary server started
2017-12-06 19:48:22,758 - octoprint.plugin.core - INFO - Loading plugins from /home/titi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.5-py2.7.egg/octoprint/plugins, /home/titi/.octoprint/plugins and installed plugin packages...
2017-12-06 19:48:23,748 - octoprint.plugin.core - INFO - Found 8 plugin(s) providing 8 mixin implementations, 5 hook handlers
2017-12-06 19:48:23,821 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/titi/.octoprint/uploads...
2017-12-06 19:48:23,938 - octoprint.filemanager.storage - INFO - ... file metadata for /home/titi/.octoprint/uploads initialized successfully.
2017-12-06 19:48:24,026 - octoprint.plugins.softwareupdate - INFO - Loaded version cache from disk
2017-12-06 19:48:24,058 - octoprint.util.connectivity_checker - INFO - Connectivity changed from offline to online
2017-12-06 19:48:32,151 - octoprint.util.pip - INFO - Using "/home/titi/OctoPrint/venv/bin/python -m pip" as command to invoke pip
2017-12-06 19:48:39,835 - octoprint.util.pip - INFO - Version of pip is 9.0.1
2017-12-06 19:48:39,851 - octoprint.plugin.core - INFO - Initialized 8 plugin implementation(s)
2017-12-06 19:48:39,866 - octoprint.plugin.core - INFO - 8 plugin(s) registered with the system:
|  Announcement Plugin (bundled) = /home/titi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.5-py2.7.egg/octoprint/plugins/announcements
|  Core Wizard (bundled) = /home/titi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.5-py2.7.egg/octoprint/plugins/corewizard
|  CuraEngine (<= 15.04) (bundled) = /home/titi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.5-py2.7.egg/octoprint/plugins/cura
|  Discovery (bundled) = /home/titi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.5-py2.7.egg/octoprint/plugins/discovery
|  Enclosure Plugin (3.51) = /home/titi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_enclosure
|  Plugin Manager (bundled) = /home/titi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.5-py2.7.egg/octoprint/plugins/pluginmanager
|  Software Update (bundled) = /home/titi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.5-py2.7.egg/octoprint/plugins/softwareupdate
|  Virtual Printer (bundled) = /home/titi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.5-py2.7.egg/octoprint/plugins/virtual_printer
2017-12-06 19:48:39,883 - octoprint.server - INFO - Reset webasset folder /home/titi/.octoprint/generated/webassets...
2017-12-06 19:48:39,913 - octoprint.server - INFO - Reset webasset folder /home/titi/.octoprint/generated/.webassets-cache...
2017-12-06 19:48:40,425 - octoprint.server - INFO - Shutting down intermediary server...
2017-12-06 19:48:40,559 - octoprint.server - INFO - Intermediary server shut down
2017-12-06 19:48:40,587 - octoprint.events - INFO - Processing startup event, this is our first event
2017-12-06 19:48:40,589 - octoprint.events - INFO - Adding 1 events to queue that were held back before startup event
2017-12-06 19:48:40,594 - octoprint.filemanager - INFO - Adding backlog items from all storage types to analysis queue...
2017-12-06 19:48:40,616 - octoprint.filemanager - INFO - Added 0 items from storage type "local" to analysis queue
2017-12-06 19:48:40,720 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from disk, was still valid
2017-12-06 19:48:41,260 - octoprint.plugins.pluginmanager - INFO - Loaded notice data from disk, was still valid
2017-12-06 19:48:41,801 - octoprint.server - INFO - Listening on http://0.0.0.0:5000
2017-12-06 19:48:41,802 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2017-12-06 19:48:41,807 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance on octopi for SSDP
2017-12-06 19:48:41,831 - octoprint.plugins.enclosure - INFO - sensorType dht
2017-12-06 19:48:41,833 - octoprint.plugins.enclosure - INFO - Setting GPIO mode to BCM
2017-12-06 19:48:41,928 - tornado.access - WARNING - 401 GET /api/printer (10.0.0.15) 49.39ms
2017-12-06 19:48:42,015 - octoprint.plugins.enclosure - WARNING - An exception of type ValueError occurred on readDhtTemp. Arguments:
('need more than 1 value to unpack',)
2017-12-06 19:48:42,074 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from disk, was still valid
2017-12-06 19:48:42,098 - tornado.access - WARNING - 401 GET /api/job (10.0.0.15) 60.84ms
2017-12-06 19:48:42,623 - octoprint.plugins.pluginmanager - INFO - Loaded notice data from disk, was still valid
[... cut ...]
2017-12-06 19:48:50,466 - octoprint.server.util.sockjs - INFO - New connection from client: 192.168.1.1
2017-12-06 19:48:50,869 - tornado.access - WARNING - 401 GET /api/printer (10.0.0.15) 40.69ms
2017-12-06 19:48:50,914 - tornado.access - WARNING - 401 GET /api/job (10.0.0.15) 19.06ms
2017-12-06 19:48:52,023 - octoprint.plugins.enclosure - INFO - sensorType dht
2017-12-06 19:48:52,203 - octoprint.plugins.enclosure - WARNING - An exception of type ValueError occurred on readDhtTemp. Arguments:
('need more than 1 value to unpack',)

vitormhenrique, 2017-12-06:
I forgot that you need to manually change one line on __init__.py file

line [41](https://github.com/vitormhenrique/OctoPrint-Enclosure/blob/master/octoprint_enclosure/__init__.py#L41) change it to **false**, save, restart, and do it again....

Just change using nano for example:

nano /home/YOURUSER/OctoPrint/venv/lib/python2.7/site-packages/octoprint_enclosure/__init__.py

ThierryDi, 2017-12-06:
Got it !

2017-12-06 20:29:17,048 - octoprint.plugins.enclosure - INFO - Temperature dht cmd: sudo python /home/titi/OctoPrint/venv/lib/python2.7/site-packages/octoprint_enclosure/getDHTTemp.py 22 4
2017-12-06 20:29:17,166 - octoprint.plugins.enclosure - INFO - Dht result: 
2017-12-06 20:29:17,168 - octoprint.plugins.enclosure - WARNING - An exception of type ValueError occurred on readDhtTemp. Arguments:
('need more than 1 value to unpack',)

The sudo Python … can’t work on my system because my user as password enabled for sudo.

Where in the code can I change the command to remove sudo ?

Thanks in advance
Thierry

vitormhenrique, 2017-12-06:
Just do a search and replace "sudo python" to "python" on this entire file....

ThierryDi, 2017-12-06:
Nailed it! thanks for your help. might be worth documenting in case some other geek does its own octoprint install

vitormhenrique, 2017-12-06:
I'll create a "force sudo" checkbox on the advanced settings on following releases, if you uncheck it would fix issues like this by not trying to use sudo.
Also, if I update the plugin it would "break" your installation. So check for this box on following releases.

ThierryDi, 2017-12-07:
I think that would be perfect

Thanks again for your support!

 

L'absence de virus dans ce courrier électronique a été vérifiée par le logiciel antivirus Avast.
https://www.avast.com/antivirus

vitormhenrique, 2018-03-01:
I just finished re-writing most of the plugin and implemented this requested feature, tested 95% of the plugin.

If you want to test it out uninstall the current plugin, go to the plugin manager, select get more, and where it says "from URL" use the development branch:

https://github.com/vitormhenrique/OctoPrint-Enclosure/archive/dev.zip

More documentation here:
https://github.com/vitormhenrique/OctoPrint-Enclosure/tree/dev

I'll push this to production after more people test it out.

d3arts, 2021-07-01:
Excuse me I have similiar problem with DHT11 - "sudo python" isnt working but "sudo python3" return temp and humidity. Thak you for advice.